grbl/grbl #944: laser power modulation by speed (and S)

felixa4, 2016-03-30:
hello dear international friends, I am italian and new in the forum and I ask you to help with a problem of all those who use the CNC laser: during acceleration and deceleration the laser burns too much or pit material; I'm not good in programming and I ask you to change grbl code (dedicated to the laser) to proportion the laser power at speed during acceleration and deceleration, for example starting with P = 1 / k (programmable via $) to speed 0, and P = 100% at full speed. on the contrary for the deceleration ramp. this is my idea.
excuse me for my limited english!!!
many thanks for your interest, byyyyyyyyyyyyeeeeeeeeee

chamnit, 2016-03-30:
@felixa4 : Yes, this is something I'm looking into. It's not difficult to implement with how Grbl's planner is designed.

closed 2016-03-30 by chamnit

felixa4, 2016-03-30:
Hi chamnit, i'm happy that you are interested to; how can i help you?
bye

felixa4, 2016-03-31:
Well, i have tested this grbl, and i see that:
the compensation is enabled for $28=0 or 1, 10, 20.....;
compensation works fine for up ramp, not for down ramp;
compensation works only when it starts from laser off, not from stop-restart ;
bye

Here is the zip https://www.dropbox.com/s/r7led3httuxwaqx/grbl.zip?dl=0