grbl/grbl #945: Hold on a Laser machine should stop spindle

BlueHairBob, 2016-03-30:
On a laser machine, when HOLD is triggered, we need the spindle to stop after all motion has stopped.  On RESUME, we need the spindle to return to the state it was in before the hold.  

As implemented right now, the HOLD causes the motion to stop - and the laser to continue burning at the stop point.

It might be nice to have a USE_LASER_SPINDLE define in the config.h

chamnit, 2016-03-30:
@BlueHairBob : Try using the safety door command. It works like a feed hold, but shuts off the spindle and re-enables it upon resuming.

An official laser mode is coming shortly. The code is complete, but needs testing and debugging. Have lots of things on my plate right now.

closed 2016-03-30 by chamnit

BlueHairBob, 2016-03-30:
That sounds great.  However, I am trying to do this from a hardware button.  The Arduino Nano has a Feed Hold on pin A1.  I see that the pin is shared with Safety door and the comment says it is enabled by a define.  I found the #define and it looks like it will work great.  I will test it today.  Thank you so much.

BlueHairBob, 2016-03-30:
OK, I tested it and it does exactly what I was hoping for.