grbl/grbl #947: Grbl Homing Issue (0,0 at Front/Left) - v0.9j

lopezj1, 2016-04-01:
Been having some issues with homing to front/left position as 0,0.  Followed Andrew Spurgeon's instructions in following link: [http://www.shapeoko.com/forum/viewtopic.php?f=4&t=7617].  I reached out to Andrew, but he was not sure on how to help.  I have limit switches on **both the positive and negative** sides of each axis (except Z).  Was hoping I could get some help here.

I have tried with both commenting & uncommenting "#define HOMING_FORCE_SET_ORIGIN in config.h" to no avail.  Here is a link to the limit switch kit I am using: [http://additive-aerospace.myshopify.com/products/shapeoko-3-limit-switch-and-probe-kit?variant=6754244737]

When I change $23 back to =0 homes fine in back right corner, but switching to $23=7 it will home properly on the X (left) but not on the Y (front).  The Y Axis motors continue moving even though the Y axis limit switch has already made contact.

I am going to test that the limit switch is working as first step, but would appreciate any other input.  

Thanks,

Joshua﻿

lopezj1, 2016-04-04:
bump

please help

chamnit, 2016-04-04:
@lopezj1 : It sounds like you have your coordinate frame misconfigured. You'll need to alter your invert direction mask on your steppers, so that the machine's cartesian coordinate system is oriented correctly. Then you can try messing with the homing settings, which should be $23=3 for x-y in the front-left and z up.

Also, make sure you enable the LIMITS_TWO_SWITCHES_ON_AXES option. This will prevent Grbl from starting a homing cycle when the limit switches are active. It's impossible to tell which side of the axes is triggered, since two limits share one pin.

lopezj1, 2016-04-05:
Issue resolved, was poor connection on front (negative) y axis limit switch.  Successfully changed home position to front left.  thanks chamnit

closed 2016-04-05 by lopezj1