grbl/grbl #948: unexpected movement during operation of the program.

Jo937, 2016-04-02:
Hi!  runtime machine suddenly starts to move is not on the program. often moves Y to zero, and then returns and continues. There are cases when the machine just changes the position where the back and continues to execute the program code G. I do not like I can not understand what it can be connected? in the executive program does not exactly have these commands I checked. but I have a screenshot of the error. I tried different programs send the code to GRBL. GRBL my version 0,9j
after the machine began to move the wrong way, I pressed the emergency stop button. See screenshot at the link below  https://yadi.sk/i/kUpgj5a_qgW9G

Jo937, 2016-04-02:
can this problem has been in the version of 3 March 2016? I will set the new look as it will behave

109JB, 2016-04-02:
I looked at your screen shots.  

What version of arduino are you using and do you know if it uses a CH340G USB to serial chip. Is your Arduino a genuine one or a copy.

Since the error happens at the same time that a status request is sent/received i am thinking it could be related to the serial stream and the problems identified with the CH340G chips in particular

Jo937, 2016-04-03:
I using Arduino UNO r3 chip CH340  and Altmega328P . I use a Chinese clone. This error occurs about once an hour or once every two hours. sometimes after 5 seconds after the start of operation after a sudden stop. Do you think it is necessary to change ARduino UNO?  and yet I use CNC Shield V3 and DRV8825? but that there is not exactly where.

Jo937, 2016-04-03:
I'm looking for on the internet but I can not find anybody that problem. but here I found http://esbuzz.net/trends/watch/vid88N6aoacx8ruw driver in which there is a version for 64-bit windows can challenge this? I will be tested over the weekend.

109JB, 2016-04-03:
I believe that the issue you are having is related to what I found a while ago in this issue #845 .  To summarize the problem, the USB to serial chip has a problem that causes the lines being sent to GRBL to be corrupted.  It happens when the chip is trying to do both send and receive functions at the same time. It is a very infrequent problem, unfortunately, the problem is related to the firmware on the CH340G chip and there is no fix for that chip yet, and there may never be one.  

The genuine Arduino UNO and some of the clones use an Atmel 16U2 USB to serial chip and this chip performs better. In delivered form, there is still a slight problem, but another user re-wrote the firmware for the Atmel 16U2 chip to completely eliminate the problem.

So, in a nutshell, you will need to ditch the Arduino clone with the ch340g chip and get one with the Atmel 16U2 chip. Then, you will need to flash the 16U2 with the new firmware from issue #845 to completely eliminate the problem.

109JB, 2016-04-04:
I updated the wiki page for known bugs to describe the issue and solutions.

https://github.com/grbl/grbl/wiki/Known-Bugs

Jo937, 2016-04-04:
many thanks. so it turns out that even the original board Arduino Uno is not saved? 
I now have a choice of two other board and I do not know what to choose.
16U2 old version:http://ru.aliexpress.com/item/Free-shipping-keyes-UNO-R3-MEGA328P-ATMEGA16U2-for-Arduino-Compatible/32298785250.html?spm=2114.10010208.100010.10.vt7aGM
a new version:http://ru.aliexpress.com/item/High-quality-UNO-R3-CH340-MEGA328P-for-Arduino-UNO-R3-NO-USB-CABLE-UNOR3-CH340/32248072684.html
what to choose?

AlexHolden, 2016-04-04:
The 16u2 version is more reliable than the CH340 version. Note that in my experience with the 16u2 there isn't much of a problem at 115200 baud or slower, it's when you try to go faster than that it can't keep up and corrupts the data transfer.

Jo937, 2016-04-04:
my links removed, in China there are two versions of the board with 16u2. in the new version in addition to the new chip is 16U2  atmega328P-PU/ What do you think, is it worth taking a novelty?

Jo937, 2016-04-04:
**109JB** You have done a great job on the study of this error, I still read. I would like to add other kinds of mistakes, sometimes performing the program simply stops, the machine stops and is not doing anything, but if you press the RUN button is a work in progress, it is necessary to upostoyanno prietom button hold.

Jo937, 2016-04-05:
I decided to join the search for a solution to this problem. I understand that the old version 0.8 had no such problems. can reduce the transmission rate of up to 9600? Now there are options besides GRBL settings.#define BLOCK_BUFFER_SIZE 18  #define SEGMENT_BUFFER_SIZE 6 #define LINE_BUFFER_SIZE 80

Jo937, 2016-04-05:
1 what I found. in device driver settings (screenshot via the link) transmission speed is set by default to 9600, in the setting GRBL set the same speed. I'll test on Thursday or Friday. perhaps this baud rate is optimally set by the manufacturer.https://yadi.sk/i/Pxq1737dqm7DZ

Jo937, 2016-04-08:
yesterday held a test at 9600 bitrate. 2 hours 20 minutes and more than 200 thousand lines of code have been transferred without a single mistake! Today bluetooth module connected to control, so I will not change speed, Bluetooth operates at 9600 all the luck.

109JB, 2016-04-08:
@Jo937 I hope this works for you, however, I did testing at 9600, 57600, 115200, and 250000 and a few of the other baud rates and found errors at all baud rates. The problem is that the errors don't always occur at the same point in the g-code program.  I tested using 1,000,000 line long g-code programs and sometimes the error would not show up, sometimes it would show up very early, and other times the error didn't show up until 800,000 lines into the program.  Because you got 200,000 lines in does not mean the problem is solved. It may be that 9600 baud is less prone to the problem, but I can pretty much assure you that it has not eliminated it.  

I am not sure what type of machine you are using it on, but I have a metal cutting milling machine and an error induced crash could result in several hundred dollars in damage to my machine. In my opinion, I would not run my machine with ch340G equipped Arduino. There are Arduino clones available with the 16U2 chip for less than $5.  I would not risk my machine for $5. Unless you can accept an occasional error induced crash, I would strongly recommend a 16U2 equipped Arduino that has had the firmware updated to the firmware that @AlexHolden modified.

I bought 4 of the 16U2 equipped Arduino Uno clones on e-bay for less than $15 shipped for all 4 of them. All you have to do to find them is search on E-bay for "Arduino Uno 16U2" and they will pop up.

Jo937, 2016-04-09:
Now I have a laser engraver but after a month I'll set the Z-axis. yes, uncertain movement router can disable it. I did spend a few tests, but believe all the previous launches were stable exact same mistake, I could even count the time until an error, the machine was able to work a maximum of 1 hour 20 minutes at the speed of 115,200. 9600 no hesitation more than 2 hours ... I had a little time to finish the code, he kept 400 thousand lines. I've got the code 800 thousand lines, this code had about 6 errors. I will spend more tests next weekend, besides I now have round, I can connect via bluetooth, the way you do not have a stable program to control GRBL with android?

Jo937, 2016-04-15:
yesterday made two file by 170 thousand lines of code. not one mistake .. Then put the third, and walked, left the machine unattended, after 4 hours approximately failed! But not in the controller but in the windowse apparently windows rebooted itself, and gave the message that windose impeccably renovated after an unexpected closing, while he remained in the machine operating position.

Jo937, 2016-04-15:
Try to replace the driver on the PC for this https://yadi.sk/d/KFI5leRwr39eg I am before all the tests changed the driver on your computer, just in case.

Jo937, 2016-05-05:
started working via Bluetooth module HC-06. using this software https://play.google.com/store/apps/details?id=org.kasdev.cnccommander . everything works fine but slow, in the settings there is a function of smart streaming protocol  when it is turned on, the machine speed increased but the stability is lost, periodically GRBL responds that in the code found an error or an invalid line of code ID25/ if you disable this function the machine was able to work 12 hours without a single error.

chamnit, 2016-05-05:
@Jo937 : The error code 25 isn't anything that would cause motion to stop or make anything unstable. It's a g-code error when there is repeated word in the line. For example, ``G1X0.00Y1.00G1`` has two G1's in it. If Grbl is throwing this error, it's a problem with your g-code program itself, or the sender or transmission of data is corrupting the line. 

If it helps you diagnose the problem, Grbl will echo back what it receives if you enable that feature in config.h. I think it's called REPORT_ECHO_LINE. Just be aware that this will cause Grbl to slow down quite a bit. You may want to increase the TX buffer size to 128 to help improve performance.

Jo937, 2016-05-05:
maybe, but you're wrong. I ran the same code 3 times and the error occurs in different places of the execution error in the program the sender for Android the link above was if you turn off the acceleration Protocol, then there are no errors at all.

lalo-uy, 2016-05-05:
May be the wireless protocol resend corrupted packets incorrectli and
generate the duplication of some words

chamnit, 2016-05-05:
@Jo937: I would like to think that I know what I'm talking about when discussing matters of Grbls source code, as I'm the developer of it. There is no reason for Grbl to respond with an error 25 unless there was a repeated word in the gcode line. That's the only time this happens. 

Like I said before, something is happening between Grbl and your sender and its most likely a corruption problem or a bug in the sender. This is not the first time something like this has happened and its almost always for the reasons above. Please try using Grbls echo line feature to help yourself diagnose where the problem is. Given the fact that Grbl runs on tens of thousands of machines without issue, it's not likely Grbl.

closed 2016-09-03 by chamnit