grbl/grbl #949: Spindle / Relay Control - M3/M4/M5

PCPayne, 2016-04-02:
Hello, I'm having issues with activating the spindle control. I'm using grbl 9, Arduino Uno, GRBL Controller 3.5.1 I can control the axis x & y don't have a z. But when I click to activate the spindle nothing happens. But if I manually type M4 it turns my relay on, then if I type m3 it turns it off. 

Issue is that the GUI Software uses M03 for on & M05 for off, is there a way for me to change this in the code?

chamnit, 2016-04-02:
@PCPayne : A recent update changed Grbl's behavior to require an 'S' spindle speed value that is non-zero to activate the spindle. Make sure that this isn't what's messing things up. Otherwise, everything works as it should and is likely something on Grbl Controller's side.

PCPayne, 2016-04-02:
Thanks @chamnit for your reply, how would I go about setting this speed?

Jo937, 2016-04-02:
try the command m3 s1000 and m3 s0

PCPayne, 2016-04-02:
No still nothing :(

Jo937, 2016-04-02:
what version GRBL and which pins to connect using?

PCPayne, 2016-04-02:
I'm using 0.9j and the relay is connected to pin 13

Jo937, 2016-04-02:
wrong. Now download this firmware. http://pastebin.com/MfWhR9Lk

chamnit, 2016-04-02:
@PCPayne : Please use the link on the Grbl front page. Also look at the Wiki to see how to wire the spindle correctly. Pin 13 is spindle direction. Pin 11 is variable spindle enable and spindle speed.

closed 2016-04-02 by chamnit

PCPayne, 2016-04-02:
@chamnit I have used the hex file as you said (http://bit.ly/1I8Ey4S) and uploaded to the uno and moved the relay to pin 11. 

But still isn't working :/

jhosorios, 2017-02-24:
Hello, My Name is Jorge. I've Just try to reconfigure a CNC to a better performance and updated the firmware from Grbl0.8 to 1.1f. Everything it´s OK, but now I have an issue (using Candle to send gcodes):
whit the grbl0.8 and Universalgcodesender the SpnEn pins used to worked ok to turn on/off the spindle using a relay. But whit the new software those pins just stay on High. How can I automate this again?

vMeph, 2017-02-25:
the CNC shield v3 was done for versions GRBL 0.8, for versions GRBL 0.9 up the pins 11 and 12 of arduino was swaped, to make it work with your cnc shield V3 one way to do it is bend the pins 11 and 12 of the shield and the solder to cables on bottom of arduino pins 11 an 12 and swap them to the conections of the shield bended pins with a female header

look at wiki schematics of arduino to see the diferences

taltoscsiko, 2017-03-11:
I'm having the same issue since I updated to grbl 1.1: Spindle enable and cooling enable pins will stay high no matter what kind of commands I'm sending. I'm using CNC-shield 3.0 with swaped pins 11 and 12 (worked fine under grbl 0.9)

Mohamed-Ateef, 2017-12-11:
Guys .. 
i Have Laser With Board control " 3 wire " " VCC - GND - PWM " 
im using CNC shield and Arduino Uno .
so im connect laser Power Supply 12 V " VCC + GND " and i Connect " PMW Wire " From Laser to CNC shield ( i Have been Tried ( Z+limit / Z-Limit / Spin En / Direction ) and nothing Happened Stepper Motor works Fine But laser Not turned On / Off 
im using Grbl 0.9J 
can u help me guys ! please

whitetd, 2017-12-11:
Update to 1.1 that supports laser power control

Mohamed-Ateef, 2017-12-11:
Can u Give Link ! 
So im correct !! My Wires is connect Correct ?! 
The issues from grbl 0.9j?

whitetd, 2017-12-11:
https://github.com/gnea/grbl

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

whitetd, 2017-12-11:
> https://github.com/gnea/grbl
>
>
> <https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon> Virus-free.
> www.avast.com
> <https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
> <#m_-1879787901572087912_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

When laser mode is enabled, Grbl controls laser power by varying the *0-5V*
voltage from the spindle PWM D11 pin.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

Mohamed-Ateef, 2017-12-12:
@whiteted
i Have been Try New grbl and control laser i fail 
i have been try (" Spin EN / Z+ / Z- / Spin Dir ") all Of that not working 
But i Measure The Voltage From (" Z + / Z- ) when the machine working with Gcode i got Size of Voltage between 2.3 MW to 3.5 V .. and i have been Measure The Voltage from (" Spin En ") it`s Give me Stable 4.99 V Not Moving Like (" Z+ / Z- ")  
idk why !! 
can u help me to make it work ?
if there any model can i connect laser to it and connect the model to cnc shield and this model can control from V ?!! 
Please Help thnx !

Jo937, 2017-12-12:
look http://3dpburner.blogspot.ru/2015/04/new-firmware-version.html

whitetd, 2017-12-12:
grbl works fine on a Eleks laser,
https://www.banggood.com/2500mW-A3-30x40cm-Desktop-DIY-Violet-Laser-Engraver-Picture-CNC-Printer-Assembling-Kits-p-1003863.html?gmcCountry=US&currency=USD&createTmp=1&utm_source=googleshopping&utm_medium=cpc_elc&utm_content=zouzou&utm_campaign=pla-us-elc2-pc&gclid=CjwKCAiAmb7RBRATEiwA7kS8VDGtWLGjW14dKyIwoXWaKLeJ3g0crlPKpicqFPYhTE1ForakFLpdkBoCfCYQAvD_BwE&cur_warehouse=CN

AnhNguyenUK, 2018-06-17:
I have the problem with command m3 s1000. At first, GRBL 0.9 auto-trigger relay to drive spindle. I use command m3 s0, but it is still triggered. However, when i use M3 S1000, the spindle turned off. After that, M3 command cannot activate relay, but M5 command can do it. What is wrong here?

Jo937, 2018-06-18:
you need to carefully read the GRBL / default settings on the spindle is fed +5 volts, on most TTL this signal indicates that you need to turn off when you give the M3 S1000 signal will be 0 volts, TTL understands this as a full turn on.

Brinkss, 2018-07-10:
I have been searching the net to find the command in GRBL 1.1 for spindle direction, can anyone point me in the right direction, thanks

Brinkss, 2018-07-10:
Thanks for the help, can you tell me where I find info on that?

langwadt, 2018-07-10:
http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m3-m4-m5

Brinkss, 2018-07-10:
Thanks for the quick reply, I guess what I would more specifically like to know is what command operates pin 13 (Spindle Direction) on the UNO

Brinkss, 2018-07-11:
Answer found, Thanks. ( The spindle direction pin D13 is a direction pin, not an on/off enable pin. M3 and M4 are CW and CCW in g-code, so M4 will turn on D13 when enabled. )

berdamc, 2018-10-19:
Hi, i am new in GRBL, i use version 1.1. I generated my gcode in Aspire 9.0, i can see in gcode the code "M3" that turn on spindle, but in the end of gcode file, don't exists M4, exists only M02. My question is, it is possible change de source code of GRBL 1.1, to make spindle off with M02 code ??

whitetd, 2018-10-20:
M3/4 are on, M5 is off

Quajetson, 2019-02-12:
berdamc - modify grbl post-processor file in aspire

109JB, 2019-02-13:
M2 in g-code DOES turn the spindle off.   If it doesn't then something is wrong with how you have it set up.

@berdamc How do you have your spindle wired to Grbl?

Quajetson, 2019-02-13:
If you dont have spindle port must use relay to control spindle circuit
(speed control thru potentiometer on power supply)

shivmahesh, 2019-08-05:
Hello there, I am having issue : error Buffer zone too small when running g code I need GRBL controller with Arduino +v3 Shield

ardmed, 2021-09-01:
> I was having the same issue trying to use a spindle relay, the spindle
> enable does still work in the new updates you only need to comment out the
> VARIABLE_SPINDLE line in the config.h file, reload the firmware and then
> the Spindle enable pin will work as normal.

Thank u