grbl/grbl #950: Homing Fail Issue

PCPayne, 2016-04-02:
Hello All,
I have been having a few issues with the homing function. I have seen on a lot of forums about this and tried everything that I have seen. 
I'm using small NO push buttons, one on each end of axis X and the same on axis Y (NOT USING Z AXIS) one wire from each button are connected to GND and the other to: Axis Y=Pin 10 / Axis X=Pin 9. Ive connected a multi meter to each of these wires and a contact is only made when the button is pressed. So deffo no short anywhere! I have tried $5=1 and still nothing, steppers don't even attempt to budge at all. Once unlocked I type $H in command line and just get: ALARM: Homing fail
----------------------------------------------------------------------------------------------
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
[Caution: Unlocked]
ok
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=3 (dir port invert mask:00000011)
$4=0 (step enable invert, bool)
$5=1 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=18 (status report mask:00010010)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=25.000 (homing feed, mm/min)
$25=750.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=6.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=100.000 (z, step/mm)
$110=1000.000 (x max rate, mm/min)
$111=1000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=5.000 (x accel, mm/sec^2)
$121=5.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok
ALARM: Homing fail
ok

Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]

ANY HELP WOULD BE GREAT THANK YOU!!

109JB, 2016-04-02:
First, your settings are wrong.
-  For normally open switches you need $5=0. $5=1 would be for normally closed switches.
-  You also need to enable hard limits by setting $21=1.  
-  Make sure the switches for each axis are in parallel since they are N.O. switches.

Also, since you are not using a z-axis, you will have to modify config.h and re-upload grbl to your arduino. About line 75 of the config.h you will see these two lines:

#define HOMING_CYCLE_0 (1<<Z_AXIS)                
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS)) 

You will need to change them to get homing to work without a z-axis.  I'm thinking replacing the two lines with this one should work:

#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))

PCPayne, 2016-04-02:
Ive done the following and just getting = ALARM: Hard Limit
(Reset to continue) 

Then after reset just repeats...

109JB, 2016-04-02:
since you are getting a "hard limit" error now, instead of "homing fail" we are making progress.

You could be experiencing noise which is causing a false limit trigger.  Do you have any filtering on your limit switch wiring.

PCPayne, 2016-04-02:
No i havnt, dont know what you mean by it. So im quessing not.

109JB, 2016-04-02:
Ok. We can do a test. Simply unplug the switches from the arduino temporarily. Then jog to the middle of the machine travel and start a homing sequence. It should start to move but you will have to execute a reset to stop it before it crashes into the limit switch. If you get movement with the switches unplugged like that then the likely culprit is noise causing a glass limit trigger. Report back and I can describe a simple filter to help with it.

PCPayne, 2016-04-02:
Ok thankyou, will give it a try in the morning & let you know how i get on.

PCPayne, 2016-04-03:
I have put a fresh install on the uno just incase ive changed something I shouldn't, But i have changed this line like you said #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)) .

Also unplugged the limit switches, once I go to run the $H the motors kick up and move slightly but then give out a high pitch noise and freeze.

I'm not sure if this is due to my other issue #952 at all.

milezdr, 2016-04-03:
Reflash the Grbl with the latest hex file on this page. Because there was a fix for the alarm status.

PCPayne, 2016-04-03:
@milezdr I have reflashed like you said, it has resolved my axis going one direction issue :)
But when I type $H with the limits disconnected. 
Nothing happens with the motors I just get = ALARM: Homing fail

109JB, 2016-04-03:
When you re-flashed, did you also change the lines mentioned in the previous post. The newly downloaded version will have the default lines in the config.h file. You will again have to edit the new config.h file.

Also, this description "... the motors kick up and move slightly but then give out a high pitch noise and freeze." sounds more like an acceleration/maximum feed rate issue. The noise you describe is what would happen if trying to drive a stepper motor too fast.

chamnit, 2016-04-03:
@PCPayne : @109JB is correct. The motors making noise is a stepper motor losing steps problem. You need to decrease the acceleration and/or max speed settings for the problem motor(s). 

Homing fail occurs in two scenarios. First, when Grbl doesn't find the limit switch within a search distance. This distance is about 1.5 times the max travel of the axis. So make sure you have your max travel settings roughly correct. Second, if the homing cycle doesn't disengage the limit switch when trying to pull off them. In this case, make sure your homing pull-off distance is enough to move the machine off the limits.

PCPayne, 2016-04-03:
Thanks for replying guys, as i uploaded the hex file. How do i edit it to change these axis settings?? 

Which numbers are the max travel distances & homing pull off $?

Also the motors move fine when jogging on speed f200 but only seem to make that noise when attempting to home. Or is there a different setting? 

Thanks again!

PCPayne, 2016-04-03:
[image]

@109JB @chamnit  As I have uploaded the hex file, how do I edit it for changing the axis settings?? 

Also what feed rate etc settings would you both recommend using? As I use F200 when jogging manually and the motors seem to handle this nicely.

109JB, 2016-04-03:
The changes to config.h need to be made and then GRBL re-flashed to the arduino. One flashed you cannot edit the config.h settings.

The machine travle limits are $130-132
The max feed rate and acceleration values are $110-112 and $120-122 respectively.

To set the optimum feed rate and acceleration I would suggest the following method.

Set both the acceleration and max feed rate to low values then move the machine around using G0 moves. If this is successful you can then start increasing the feed rate and acceleration. I would start with the feed rate first leaving the accel alone. Make a change then test. Once you reach a point where the axis stalls, back the feed rate off to about 75% of the max achieved. Then do the same for acceleration.

Since you mentioned it works well when jogging at F200, your should be able to set the max feed rate to 200 to start and move up from there.

Note that there are also feed rates for homing that are set by parameters $24-25. The homing seek rate ($25) is the first rate the homing sequence moves at and will move at this rate until it trips the switch the first time. Then it will back off of the switch by the distance defined in parameter $27, then it will again move toward the switch at the homing feed rate set by $24.

PCPayne, 2016-04-04:
Ok thankyou i will give it a try. Wont be until the weekend now though.

PCPayne, 2016-04-12:
[image]
Hi everybody, ive finally had time to try this again. BUT still no luck :( 
I have edited the confiq file to show: #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)) and uploaded it via Arduino Software. Once uploaded i unlocked via $X and then tried $H (With no Limit Switches connected) i heard a very slight murmur from the X axis motor but no actual movement. Before when i entered F200 and tried manually jogging via GRBL Controller the motors would work fine. Have i got my settings correct? As I'm getting very confused. Thanks for everybody's help !!

PCPayne, 2016-04-12:
@109JB @chamnit  By the way I'm using (17HS4401-S) Stepper motors. Don't know if that helps at all.

wslewis, 2016-12-28:
Hi, I'm having trouble homing. I've checked out all the posts and can't find a solution. Here are my settings:
[image]
I just want to point out I wouldn't be asking for help if I hadn't tried out every possible solution I can think of including updating to version 1.1 to no avail. My limit switches are shielded and grounded, also the limit switches activate properly when hit. I've tried manually activating the limit switches that are now in the front left position (because I want x, y to be in the front left position) to activate and that didn't work. I assumed it would work seeing as how evidently grbl looks for limits in the positive direction. So what's the problem???? As it stands now, although in homing mode the Z axis works flawlessly when it starts looking for x and y limit switches it just trods right on past the limit switch even though it is activated!  One more thing.. I tried compiling using arduino to no avail either. I'm on Windows 10 and using an arduino nano - could that be it?

swarfer, 2016-12-28:
So, to confirm:
The switches actually work
X+ movement is *tool* to the right of the work
Y+ movement is tool away from the work/you
Z+ movement is up, away from the work
If the actual axis directions are not correct, homing will never work.

V1.1 is not flawed, but a lot has changed, including the messages it sends back to the GUI, so you have to upgrade your GUI if you upgrade to V1.1.    What GUI are you using, and did you upgrade it?
The command descriptions had to be removed to save memory so it could be used for extra features, like jog commands and laser mode.  You really should read the release notes before moaning.

If compiling fails in Arduino IDE it usually means that you have 2 GRBL libraries installed at the same time.
You have to completely remove GRLB 0.9 from the libraries folder before you copy the 1.1 folder in, otherwise they interfere with each other and compiling fails (because the IDE tries to be far to clever about finding files automatically and will find duplicates it cannot deal with).

> "although  in homing mode the Z axis works flawlessly when it starts looking for x and y limit switches it just trods right on past the limit switch even though it is activated"

seems to me that though the switch works when YOU push on it, it does not work when the axis pushes on it.
How far does it need to be pushed in order to make connection?

wslewis, 2016-12-28:
Hey swarfer.. thanks for answering. The limit switches activate when I jog manually ($21=1) in x,y and z but I just can't get the homing system to work no matter what. Can you have a look at my settings above and see if you can detect what could be causing the problem? As mentioned, Z axis homes perfectly but when it comes to x and y despite the limit switches being activated (they are placed correctly and do get triggered) it just ignores them forcing a soft reset to stop the gantry. The GUI I'm using is the universal g code sender.

swarfer, 2017-01-03:
Since homing works for other people, the software is probably ok.
So it must be hardware....    
maybe you have the X switch connected to the Y pin?

wslewis, 2017-01-04:
thanks for answering. You could be right.. it makes sense. i'm going to check that out and will also try repositioning the switches.

wslewis, 2017-01-04:
@swarfer Problem Solved! Switching the x and y pins worked. So the way I have it now is limit-X goes to arduino pin D9 and limit-Y goes to D10. Thanks again.

jcassago, 2017-10-10:
109JB , could you describe the simple filter to help with "ALARM: Hard limit (Reset to continue) you've mentioned on 2 Apr 2016 to PCPayne. This happens when I turn on my spindle. Thank you.

aqibi2000, 2017-11-06:
@PCPayne I have the exact same issue: Hi everybody, ive finally had time to try this again. BUT still no luck :(
I have edited the confiq file to show: #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)) and uploaded it via Arduino Software. Once uploaded i unlocked via $X and then tried $H (With no Limit Switches connected) i heard a very slight murmur from the X axis motor but no actual movement. Before when i entered F200 and tried manually jogging via GRBL Controller the motors would work fine. Have i got my settings correct? As I'm getting very confused. Thanks for everybody's help !!

Did you get anywhere?

robingantois, 2017-12-23:
@PCPayne I have the same problem. Did you finally find a solution? I also have the same 17HS4401 Stepper motors.

hamarinda, 2020-01-01:
@PCPayne @aqibi2000 #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)) and 5=1
Invert the limit pin
Its work to me

AndreVoro, 2023-02-20:
Hello!
I ask for help on GRBL 0.9 + Ramps 1.4 for a three-axis project (sold in stock from an old 3D printer). It is very important for me to follow the HOME procedure along the three axes. Z should move first, then X and Y.
There are lines in the config.h file
#define HOMING_CYCLE_0 (1<<Z_AXIS) // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS)) // OPTIONAL: Then move X,Y at the same time.

But! Only HOMING_CYCLE_0 is executed

Please tell me how to make HOMING_CYCLE_1 work?

PS. The switches work great, no interference.

AndreVoro, 2023-02-20:
I tried like this
#define HOMING_CYCLE_0 ((1<<Z_AXIS)|(1<<X_AXIS)|(1<<Y_AXIS))

But it doesn't work, bounce $27=1.00 happens only on the x-axis. And I need n all axes.

m1shg4n, 2025-01-19:
If the homing function does not work for you, try removing the jumpers under the stepper motor drivers. Probably the homing function uses a different pulse duration for the motors and if you set a large step divider, the motor cannot move.

AlexTMFN, 2025-09-17:
i had the same problem. I needed a 2 axis cnc. I went to https://github.com/gnea/grbl/releases/tag/v1.1h.20190825, downloaded the zip and extracted it. Went in grbl/config.h and edited 104-118. Read the notes to understand better, i just commented line 105-106 and uncommented line 110 and 118 to get rid of z axis.

Then ran `make clean && make' in the same folder to compile it as a hex file then uploaded it via avrdude. 

Worked first try