grbl/grbl #951: Steppers stutter on PWM Spindle change

El-Tonno, 2016-04-02:
Hi all,

just builded a Laser Cutter / Engraver. Standard setup Arduino UNO / CNC Shield / Pololu A4988.
Everything is fine but the engraving with varying "Spindle Speed" or better varying Laser intensity stutters.
Seems to be that each 'Sxxx' command causes a short break in Stepper movement.
Or better i think each 'Sxxx' forces a 'M03 / M05' which causes the break. Also tested with Coolant control M8 and M9 instead of M3 and Sxxx... same issue.
Tested several Gcode probes with M3-M5 / M8-M9 also without M3* commands and just spindle change commands at Spindle in OFF state. No stuttering! .. But had a look at spindle_control.c -->
Of course there is no stuttering (if (state == SPINDLE_DISABLE))
Spindle speed change does nothing in OFF state :-(

I was not sure if this could be an hardware issue (PWM change could cause some interferences on board and also the stuttering). But i think the test with M8/M9 proof that this is an software issue. 

Is there a solution for this?
Couldn`t find the place where Sxxx sends M3 or M5.  
Or is it just that there are no parallel tasks for voltage change of PWM output and the procedure of speed change needs the time which causes the stutter.
Would be very nice if spindle speed change or coolant control can be changed in realtime without affecting the smooth movement.

Please help me to understand whats wrong here. 

Thanks a lot :-)

tklus, 2016-04-02:
I had a similar issue and ended up burning the jtech grbl 9g firmware and setting the accel rate on X and y to 5000. Just google search jtech grbl 9g. The current v 9j has a pause for spindle speed change and that causes the jerkiness.

http://jtechphotonics.com/?page_id=1100

tklus, 2016-04-02:
http://www.picengrave.com/forum/viewforum.php?f=30&sid=2ab503a082778005dd23e445d8322ff7

http://www.picengrave.com/forum/

chamnit, 2016-04-02:
@El-Tonno : Grbl doesn't officially support laser cutters and is designed for CNC mills and routers, where you have to stop to perform a spindle speed change. The current solution is to use JTech, but Grbl will be supporting lasers in the very near future.

closed 2016-04-02 by chamnit

El-Tonno, 2016-04-03:
@tklus .. Thanks a lot !!!... works fine ... tried a little until i found out 0.9g has $14  auto cycle option ;-)

tklus, 2016-04-03:
Great! Glad to help

El-Tonno, 2016-04-03:
@chamnit  ...sorry for this unneeded thread ... but i was not sure if this is a bug or missing feature or just a hardware issue. would be nice if this feature is added soon:-) 0.9g works fine for me but its precompiled HEX... so simple changes for my machine are more complicated.. looking forward for your efforts ... thanks :-)

tklus, 2016-04-03:
I am also excited about a built in laser mode. I currently have to reflash firmware if I want to use my spindle and then back to 9g for the laser mode.

Thank you all for the hard work put into this :D
Tim

admelyssis, 2016-05-22:
@chamnit ... I just arrived at the same point and wonder if 0.9g is still the best option or if there is already a development branch with a solution worth testing? If not - maybe you could suggest a line in the current version where the "stop on spindle speed change" can just be commented out?

chamnit, 2016-05-22:
@admelyssis : Yes. It's coming very soon in v1.0. Please be patient. It's not a trivial fix to allow continuous spindle speeds as you suggest. It involves new code that passes spindle data through the planner to the step execution, and that code is integrated with the new feed/rapid/spindle overrides in v1.0.

admelyssis, 2016-05-23:
@chamnit ... Thanks for the fast response! Great to hear, that it is planned for the upcoming v1.0 :) 
p.s. nice source code

splitn2, 2016-05-24:
v1.0 :) I can't wait the suspense is killing me !!!!! thanks again to @chamnit for all his hard graft keep up the great work mate.

chamnit, 2016-05-29:
@admelyssis @splitn2 : I know. It's been a long time coming. But don't worry, most of the major GUI developers have pre-release versions to get their GUIs updated before the official release. So, users will be able to start using it almost immediately.