grbl/grbl #952: X AXIS ONLY GOES ONE DIRECTION

PCPayne, 2016-04-03:
Since I have updated to the newest version on the code screen 0.9j when I move the axis via jog the x axis only moves to the left. Y axis is fine and moves both ways.

Any ideas?

PCPayne, 2016-04-03:
Issue Resolved - If I uploaded the grbl file via Arduino the axis would only go one way.
Once I uploaded the HEX File below, issue has been resolved.

closed 2016-04-03 by PCPayne

109JB, 2016-04-03:
The method of uploading GRBL to the arduino should not make any difference. The Arduino IDE or HEX upload are simply two different methods for the same end result.

PCPayne, 2016-04-03:
@109JB I understand they are the same way to upload, but how do i edit the hex file? As i created the hex file via copying the hex, making a text document and using xloader to upload. 

But using IDE i can edit the file in the arduino libary then upload that way. 

Or do i flash with hex using xloader then upload grbl via IDE. Im a little confused sorry!

109JB, 2016-04-03:
If you use the Arduino IDE, you can edit the human readable program files, like config.h, and when you upload using the IDE, it performs all of the necessary steps to create the hex file that is uploaded to the board. You do not directly edit the hex file. You edit the source files and the IDE does the rest

PCPayne, 2016-04-04:
So just so i understand it right, here on the grbl page there isnt any differance between the hex file download to the zip file. The contents are the same. Its just you can edit the zip file download?