grbl/grbl #953: lost connection, with Port, no feedback  from adruino

julienja, 2016-04-05:
Hi, I am new to this world
I bought a kit on dx.com from NeJe, I have the grbl controller 3.6.1
Adruino Laser Cube.

When I wanted to reposition the laser via the axe commande, the system jammed, the y axis motor doesnt move and getting really hot. I cannot connect to the adruino via the GRBL controller.
Message:
No data from Serial Port, waiting from Gbrl controller

Of course I tries different rebout, but when I replug the y axis motor is getting hot.
I reinstalled the port, the computer react when plug and unplugged
Is there another to access the adruino and reintall the hardware?

Thank you for your help

chamnit, 2016-04-05:
@julienja : This isn't a Grbl problem, rather it's a problem with your board or wiring. It's likely your stepper wiring is incorrect or your board is fried. Please contact the board manufacturer for help.

closed 2016-04-05 by chamnit

julienja, 2016-04-05:
Do you mind to be a bit more specific, how I can i make the good diagnostic?
the wiring seems to be good, How can I be sure is the board fried?

chamnit, 2016-04-05:
@julienja : Disconnect everything from your board. Only have the bare minimum to communicate with Grbl. If you can't connect, you may be out of luck. IF you can, add one thing at time to determine the problem wiring setup.

julienja, 2016-04-05:
Argh I am out of luck, it worked only 3 days! I did few little engraving and respected the cooling time.
If I cannot  get a new board, what do you recommend? and how to install it?
thank you to be so quick