grbl/grbl #954: Some equations of motion for steppers

Moffy, 2016-04-07:
I have been trying to understand the GRBL motion code, so I went back to v0.6, because it was a bit simpler. I also had to work out the basic motion equations so that I could understand what is being coded. I have attached my cheat sheet:
[motion.pdf](https://github.com/grbl/grbl/files/209396/motion.pdf)
The thing I am not sure about with GRBL is how it works out the stepper timing. It looks like the acceleration is calculated as: Fnew = Fold + DeltaF(acceleration), F is frequency, Is that correct?

Moffy, 2016-04-09:
Seem to have found the explanation: http://picprog.strongedge.net/step_prof/step-profile.html

Moffy, 2016-04-12:
Did a simulation timing using the Keil simulator for ARM Cortex M3 for equations 6&7 and got: 3.17us, that's for non optimised code. The great thing about the DUE and ARM cores is that you can prioritize interrupts. A higher priority interrupt will interrupt a lower priority one. You can make the Stepper.c ISR the highest priority, which just outputs the step pulses and updates the timing, and it then calls a lower priority software ISR to do all the calculations for the next pulse, time. Means you can have microsecond accurate pulses.

Moffy, 2016-04-14:
Just for completeness I simulated equation 6 on an ARM Cortex M4 with a floating point unit: 333ns. When I optimised for the M3: 2us. They are pretty fast. Clock in both cases was 84MHz.

J-Dunn, 2016-04-23:
Yes , one of the drawbacks with Uno is that the USB port gets higher priority INT than the steppers, which is pretty backwards. This is why you can not interrogate the various parameters when the machine is in motion because the serial IO could end up in a wait loop and mess up the motor control. 

ARM Cortex h/w is about the same price as a UNO and at least and order of magnitude more powerful in just about all respects.

chamnit, 2016-04-25:
@Moffy: Thanks for the test results. I've been coincidentally working on the next planner, and the math required does seemingly require something like an Cortex-M4, rather than an M0. Especially, if I'd like to have a bit of breathing room for future advances.

For the most part, Grbl's limitations aren't the interrupts or priorities. It's the speed of which the little 8-bit AVR processor can compute everything. Rather than nano-seconds, computations like sqrt() or cos() takes upwards to 0.25 - 0.4ms each. That's almost absurdly slow. I'm actually surprised that even with this, Grbl is still able to be serviceable for most applications.

closed 2016-04-25 by chamnit

J-Dunn, 2016-04-25:
Yes, doing trig functions on hardware like that is very slow. 

GRBL has some clever optimisations to get all the processing done and manage to retain a respectable pulse rate. It's surprising that it does as well as it this on such limited hardware and even have a little code space left free. 

I was glad to be able to squeeze a bit of SPI in there too for the L6474 stack, thought that pushed it to the last few bytes of flash. 

Having the  serial interface as the highest priority is a bit unfortunate. Luckily we now have a huge choice of inexpensive hardware.

andywang1005, 2017-01-11:
@Moffy how did u do the simulate of ARM Cortex M3 by using the Keil? could u give me some tips or codes? PS: I am a beginner :)

Moffy, 2017-01-12:
I have attached my project files from Keil. I don't use it any more so cannot offer further advice. Hope it helps.
[Keil.zip](https://github.com/grbl/grbl/files/703124/Keil.zip)

andywang1005, 2017-01-16:
Hi Moffy,

Thank you very much!

Best Regards

Andy