grbl/grbl #955: Right shape wrong size

GDC01, 2016-04-09:
If anyone here could help me i would greatly appreciate it. I have a shapeoko 2 and I just down loaded and flashed the latest version of GRBL. First I had to update the arduino program thing, I thinks its called an IDE? Anyways I got it flashed and was testing it out, my Y and Z were backwards but it was no problem, I just reversed the polarity on the stepper motors and now everything is moving in the right direction. My problem is that it moves/cuts way way too big. I would say more than 5 times too big. It has the right shape but just way too big. I have tried downloading and flashing the arduino many times with no luck. I would imagine that there is some code or something I need to change but am clueless. I was looking in the config-h file around lines 100-104. There is something there but dont know if thats what i need to change. Please someone help me. Thank you in advance.

electrokean, 2016-04-09:
You need to configure Grbl to suit your machine - in particular the steps per mm ($100-$102), but you should review all the defaults
See this article for detailed info: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

GDC01, 2016-04-09:
okay, wow. thank you so much @electrokean I figured out that they were at 250 and with some searching on the internet they are supposed to be at 40 steps per mm. They whole reason I updated grbl in the first place was because I was sick and tired of how slow it moves when transitioning between cuts. Now it is moving slow again and moves faster when i turn the feed rate up and it's cutting than when it transitions. It would appear that the solution to this is better stepper motors... Back to square 1. Thank you so much for your help.

electrokean, 2016-04-09:
You can try adjusting the max feed rate settings ($110-$112) and acceleration ($120-122) for each axis, but if your steppers or drivers aren't up to it you may not get much improvement.  Don't have any experience with the Shapeoko myself.

electrokean, 2016-04-09:
If you're able to get it to move faster by setting a higher feed rate, then in the Shapeoko software (or whatever you use to generate your G code) you may want to see if you can adjust the feed rate assigned to rapid (G00) movements.

carneeki, 2016-04-12:
Do you think there is value in creating a couple of Wiki pages to "copy/paste" config settings for various COTS kits like the Shapeoko?

Basically - people download and flash the firmware, and then paste the settings into a terminal / file once to set themselves up, and then they can make some test movements.

closed 2016-09-03 by chamnit