grbl/grbl #956: Right shape, but design off

julienja, 2016-04-10:
Hi,
When I load my design from laser engraver extension. At speed 60 in grbl controller, the design looks good. But after my engraving is off at some points 1cm too left or too right. I reworked my design adding extra bridge to help the path. My graving area is 16x4cm. My steps are 80mm/mn.

If you have a configuration for large engraving, I take it!

Thank you, it s A great little soft for everyone to engrave is own path .

chamnit, 2016-04-10:
@julienja : Grbl does not have an official laser engraver extension. These are usually supported by other developers for their products. That said, it's difficult to understand what your problem is. A physical shift in a gcode program is usually caused by improper motor and/or junction deviation settings.

closed 2016-04-10 by chamnit

julienja, 2016-04-10:
OK I check for the deviation setting,do you have  a setting recommended for this one?
It s only on few parts of the all engraving, not everything, the rest is sharp, weird?

chamnit, 2016-04-10:
@julienja : About 0.02mm is recommended for most applications. There is also a known issue with Chinese Arduinos with the CH340 USB-serial chips. (Not FTDI or atmega8U/16U found on better and official boards). The CH340's internal buffers will overflow and corrupt the data stream.

julienja, 2016-04-10:
OK I will try it was at 0.05mm!
Thank you to be so quick, so reactive , many companies should learn from you.
I cool down the laser and I let you know

Thank you

julienja, 2016-04-11:
You  damm good it s now good at 95%, do you have other setting recommendations ?