grbl/grbl #957: Is CoreXY functional in 0.9?

elbowdonkey, 2016-04-13:
I have a pretty simple setup - a corexy plotter with no z-axis. There are also no limit switches currently.

I've tried two different homing settings in `config.h`, including:

```
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)
// #define HOMING_CYCLE_2
// ...
#define COREXY
```

And

```
#define HOMING_CYCLE_0 (1<<X_AXIS)|(1<<Y_AXIS)
// #define HOMING_CYCLE_2
// ...
#define COREXY
``

Regardless of settings, the machine behaves like a standard XY bot.

What am I missing?

jaketeater, 2016-04-13:
I had a similar issue where I modified the homing cycle in config.h file and nothing seemed to happen.  It was because I was editing the wrong config.h file.  After adding the library to the Arduino IDE, it created a new copy of the source and put it in ~/Arduino/libraries/grbl/

Also, here is my working 2 axis homing cycle:  
`#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))`

Lastly, you don't have any limit switches - so the machine won't be able to home.  Have you turned off homing with $22=1?

elbowdonkey, 2016-04-13:
I'm pretty certain I was modifying the correct code. I'll ensure that there's no chance I'm editing the wrong files and give it another shot.

Regarding homing - I'll double check that as well. But doesn't `$22=1` mean "turn on the homing cycle"?

Another thought occurred to me - if my steppers aren't stepping in the right direction, it obviously won't look like CoreXY is working. I'll have to set the various permutations of directions to see what impact that has.

jaketeater, 2016-04-13:
Sorry, my mistake, $22=0 would be off.

elbowdonkey, 2016-04-14:
Still no luck here.

I verified that my simple CoreXY plotter was mechanically correct.

I verified that I was editing the correct files before compiling and uploading using the Arduino IDE (version 1.6.7). 

Here are the steps I took, all from command line (I'm on OSX):

```
cd ~
rm -Rf ~/Documents/Arduino/libraries/grbl
mkdir grbl-tmp
cd grbl-tmp
git clone https://github.com/grbl/grbl.git
```
I then opened the Arduino IDE (1.6.7), checked to ensure that there were no examples for grbl, then went to `Sketch > Include Library > Add .ZIP Library...` and navigated to `~/Documents/Arduino/libraries/grbl` then clicked `Choose`.

I verified that the grbl example was now available.

I use an external editor. This means I could then do:

```
cd ~/Documents/Arduino/libraries/grbl
atom .
```
Atom is my editor. From Atom, I opened the examples/grblUpload/grblUpload.ino file and added a comment to the last line saying `// Find me in the IDE`.

I verified that the change was made to the grblUpload Sketch in the IDE.

I then went back to Atom and adjusted `config.h` to have these homing settings:

```
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
```
And turned on COREXY:

```
#define COREXY
```

After saving, I checked to make sure my stepper drivers (Pololu A4988 carriers) were hooked up to the correct pins on my Arduino UNO by verifying that the pins defined in `cpu_map_atmega328p.h` matched the pins I was using.

I then compiled and uploaded to the Arduino.

After ensuring that homing was turned off (`$22=0`) I ran a few G code commands:

`G0 X3 Y0` should move the X and Y steppers counter-clockwise, but only the X stepper rotated (clockwise).
`G0 X0 Y3` should move the X stepper counter-clockwise, and the Y stepper clock-wise, but only the Y stepper rotated (clockwise).

`$3` was set to `0`, so I set it to `$3=1`. When given `G0 X3 Y0` the X stepper rotated counter-clockwise. The Y stepper was idle.

Similar story for `$3=2` and `$3=3` (which covers the XY possibilities while ignoring Z axis).

Even before messing with `$3` it was obvious that I wasn't in CoreXY. There's no situation with CoreXY where only one stepper moves at once.

So I'm stumped again. I will dig into the code to see if I can find anything. Before that though I'm going to load a simple sanity check test to verify that my steppers are working properly. I'll just load a simple test using the AccelStepper library.

lytke, 2016-04-28:
CoreXY works just fine with 0.9j. !!

Here is what I changed in config.h
1.
Add "//" to the below two lines to disable.
-----------------------------------------
//#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.

2.
Uncomment these two lines
#define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY specific
#define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY specific

3.
And uncomment this last line:
#define COREXY // Default disabled. Uncomment to enable.

4.
Compile and upload.

You may need to check the cables to the stepper motorors to get X and Y directions right.

minicico, 2016-05-10:
hello, i change these lines, but machine homing in X and Y at the same time.homing will stop when you hit the X or Y switch.homing does not continue to next axis.

closed 2016-09-03 by chamnit