grbl/grbl #959: Spindle frequency control

Georges760, 2016-04-15:
I intend to use grbl to control an electro-valve instead of a laser on a XY mechanic, in order to have a mono-nozzle print head.

I am quite new to grbl, but definitively not to AVR/Arduino/3Dprinting.

I understand that v0.9 of grbl can handle a variable spindle in PWM mode, this is great for my need (because it control the drop size). But I would also like to control the spindle signal frequency (not only the PWM ratio).
I need this because for drop on demand application, we need to control the number of drop per lenght unit. I can use the move speed parameter to control this "drop number per length unit" with a fix drop frequency, but it would be much more powerfull to also control the drop frequency, so the variable spindle frequency in grbl. 

So I have 2 questions : 
- is there an way with actual grbl to achieve my goal ? or do I need to code some more specific command ?
- if I need to code these new specific command, how would you do it ? I will do it, but I prefer ask an more experimented person first to follow the good path at first...

If I have to do it alone, I would add an optional extension to the "S" command for example : 
S500 100, where "S500" means the variable spindle PWM ration (over 1000) and the "100" means somehow the desired variable spindle frequency (still need to find an usefull unit for this value...).

I don't know if it will profit other people on other application, but in case I would like to do the software evolution the right way.

EliteEng, 2016-04-16:
@Georges760 I can help out with this one.
There has been a few issues related to different variations of PWM/Frequency for the spindle so I will have a think about how best to cater for as many as possible

Georges760, 2016-04-18:
After looking at code, it look like PWM is done by Timer2 in Fast PWM mode with WGM22 = 0, so the frequency is fixed by a TOP equal to MAX (0xFF). 

A way to acheive my goal is to stay in Fast PWM mode but changing WGM22 to 1, so the TOP value is now the value in OCR2A, which means OCR2A control now the frequency (not anymore the PWM ratio). In this mode, we need to use the other output compare OC2B for the PWM ratio, which means the pin will change from PB3 (OC2A - Arduino D11) to PD3 (OC2B - Arduino D3) which is already used for Y_STEP, which is a serious problem of hardware compatibility... You already had to change this pin between v0.8 and v0.9 for a similar reason... So this is possible for me to change all these pins in my prototype, but I doubt it will go main stream easily...

rasciodc, 2016-07-20:
I am using a RAMPS 1.4 Board with GRBL, which means I have a lot of pins I can use to so spindle control.

I've managed to make the code work on pins D7 (nothing connected to it but no header pin soldered), D9 (which outputs through a transistor, not what I'd like) and pin D12, but it would be perfect if I could make it work on D11 (which is an auxiliary pin on the board, with headers soldered on).

The thing is that I can't make it work on D11! All pins I managed to get to work are OCRxB (OCR2B, OCR4B and OCR1B, respectively) and pin D11 is OCR1A. Here is the pertinent part of the code:

```
#define` PWM_MAX_VALUE 65535.0
#define TCCRA_REGISTER    TCCR1A
#define TCCRB_REGISTER    TCCR1B
#define OCR_REGISTER    OCR1B
#define OCR_REGISTER_TOP OCR1A

#define COMB_BIT      COM1B1
#define WAVE0_REGISTER    WGM10
#define WAVE1_REGISTER    WGM11
#define WAVE2_REGISTER    WGM12
#define WAVE3_REGISTER    WGM13

#define SPINDLE_PWM_DDR   DDRB
#define SPINDLE_PWM_PORT    PORTB
#define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 11

#ifdef VARIABLE_SPINDLE
     
#define SPINDLE_RPM_RANGE (SPINDLE_MAX_RPM-SPINDLE_MIN_RPM)

        TCCRA_REGISTER = (1 << COMB_BIT) | (1 << WAVE1_REGISTER) | (1 << WAVE0_REGISTER);
        TCCRB_REGISTER = (TCCRB_REGISTER & 0b11111000) | (1<<CS11) | (1<<WAVE2_REGISTER)| (1<<WAVE3_REGISTER); // set to 1/8 Prescaler
        OCR_REGISTER_TOP = 0xFFFF; //This is OCR1A
                                    
        rpm -= SPINDLE_MIN_RPM;
        if (rpm > SPINDLE_RPM_RANGE) {
            rpm = SPINDLE_RPM_RANGE;
        } // Prevent uint8 overflow
        uint16_t current_pwm = floor(rpm * (PWM_MAX_VALUE / SPINDLE_RPM_RANGE)+0.5); 
        OCR_REGISTER = current_pwm; //This is OCR1B
                
#else   
        // SPINDLE_ENABLE_PORT |= (1 << SPINDLE_ENABLE_BIT);
        rampsStartSpindle();
        
#endif
```

Any thoughts on how to switch pin D12 to D11?

rasciodc, 2016-07-20:
Nevermind, got it..... Changed pwm mode to 14 to have ICR1 on TOP and free OCR1A register to use pin D11!

The output is flutuating +-0.06V though... should I implement some filter or average samples?