grbl/grbl #960: Suggestion for Z-Axis Jogging

toddl1962, 2016-04-15:
I have an X-carve controlled by GRBL.  On a few occasions I have inadvertently jogged the Z-Axis using a step size, such as 1 inch, whereas I usually only use 0.1inch or 1mm for Z-Axis jogging.  Sometimes this results in the bit crashing into my work piece.  Fortunately I have not caused any permanent damage to my machine as the maker slides are pretty flexible.  

I feel an additional edit field or spin control to set the Z max step size would be a nice safety feature.  The default could be "no max".  If the Z max step size is set then the Z axis would jog at the lower of the set step size or Z max step size.  This would be very convenient for jogging the X and Y in large steps and not risking damaging the machine with large Z steps by accident.

chamnit, 2016-04-15:
@toddl1962 : There is something already existing to prevent an axis limit crash. That's soft limits. Otherwise, the max step size thing is something controlled by the GUI, not Grbl explicitly, because Grbl only moves when its told to and how far via commands from the GUI.

However, I do plan on introducing a seperate jogging mode for Grbl that will hopefully eliminate problems like this. It depends on how much space I have left to play with v1.0.

closed 2016-04-15 by chamnit

toddl1962, 2016-04-16:
Thanks for the quick response!  Actually I do use soft limits but they have to be kind of loose for the Z axis because I don't really know the bit length I will be using.  Thinking about it this functionality would be more suited for the sender program.  But thank you for considering my request!

toddl1962, 2016-04-16:
By the way I wanted to mention GRBL is a great program.  It works very well for me and I appreciate the work you guys put into maintaining it.  I am an embedded software developer myself so I appreciate the amount of effort to maintain the codebase and manage the bug log.

jahnj0584, 2016-04-16:
this question is probably better suited for Wil who does the UGS.

mschorer, 2016-04-16:
Hi guys,

i've implemented a tool table (with tool lengths and other stuff) and soft
limits that take tool lengths into account to handle that issue.
It's still not exact, as we don't have support for a tool changer (with
exact tool length set up) ... you have to set your tool to the predefined
depth manually.
The next step would be to do tool length sensing after every tool change.

See the code @ http://github.com/mschorer/grbl