grbl/grbl #962: 4-th Axis

CNCmax, 2016-04-16:
Hey Guys,

I've got a 4-Axis CNC machine. The 4-th axis is a rotational.

Could you please tell me when there will be a update for 4 axis GRBL. 

I really like your project. Thank you!

electrokean, 2016-04-21:
Well, if you want official support ... I imagine that it will be ready when it is ready.

In the mean time, it is pretty easy to add more axes if you're running on an Arduino Mega (to get more I/O pins).  I actually got an 8-axis machine running no problem (some axis linear, some rotational), but apart from X & Y, we would only ever move a single axis at a time.

These are probably a bit (quite) out of date, but if you're willing to modify code then they will give you a starting point:
https://github.com/EliteEng/grbl/tree/6-AXIS
https://github.com/electrokean/grbl/tree/6-AXIS

cheton, 2016-04-21:
The source code is worth to read. I think it's not so hard to change to code to support more axes.

https://github.com/EliteEng/grbl/commit/e92ffa9b1d340516cd4723f15298d47008d4e475#diff-6a41552e473d809d4032ec8b77cf75b9

chamnit, 2016-04-25:
@CNCmax : I'll be adding it shortly to my separate [Grbl-Mega fork](http://www.github.com/gnea/grbl-Mega/). But, it'll be ready when its ready. I've got a lot of the things to work on first.

closed 2016-04-25 by chamnit