grbl/grbl #965: Precision for cnc machine using grbl

Dareblix, 2016-04-18:
Hello everyone before i explain this issue, well i just finished my machine, these are my characteristics:
Nema 17 steppers
Easy drivers
1/4" threaded rod
Arduino uno
Grbl 0.8 and universal g code sender
Okay, this machine works aparently good, but what i want to reach is the precision in measures, when i draw something it's not even close to the measures i drew! well i just draw a logo with inskape, but it is for my settings for sure, i saw some tutorials about how to configurate grbl and the people who use a 1/2" threaded rod sets the steps per mm with 200, im using a 1/4" threaded rod so i configured mine with 400 (this is bad for sure), i don't want full precision i know that'll never happend but i want something close to that. these are my grbl settings:
0=400 x step/mm
1=400ystep/mm
2=400zstep/mm
3=15 step,pulse, usec
4=350 default feed mm,min
5=500 default seek mm,min
6=192 step port invert mask, int 11000000
7=253 step idle delay,msec
8=999 aceleration,mm, sec ^2
9= 0.050 junction deviation =,mm
10 =0.100 arc , mm/ segment
11= 25 narc correction ,inc
12= 3, n decimals , int
13= 0 report inches, bool
14=1 autostart bool
15=1 invert step enable, bool
16=0 hard limits bool
17=0 homing cycle, bool
18 =0 homing dir invert mask , int 00000000
19= 500.0 homing feed, mm,min
20=500.00 homing seek,mm, min
21=100 homing debounce,msec
22=1.000 homing pull of, mm
i want to use catia to generate my g codes and grbl controller if i reach this, if you have any recomendation please let me know, thanks everyone for read guys!!

shmorgan, 2016-04-18:
More important, what is your thread per inch? And is the threaded rod connected directly to the stepper motor?

Dareblix, 2016-04-18:
Shmorgan thanks for answer idk about thread per inch i cannot see that command in universal gcode sender, and the motor is not directly to the rod i used a 1/4" couple to engage the rotor with the rod, it moves well

jahnj0584, 2016-04-18:
all you need to do is clamp a ruler down. Jog 100mm, and record how far it went. Do 100/(distance moved) and multiply that by your 0 and 1 settings for each axis until moves 100mm in real life when you jog 100mm. It takes me 2 jogs per axis to have it set perfectly, and since you use rod you'll never have to change it.

lalo-uy, 2016-04-18:
In orther to get your calibration, start with your actual setting of 400ppm
Jog your machin exactly 100mm and count how many steps of the machine say it has moves. Say it move x mm, the to move 100mm your machine needed 400*x steps => so your calibration is 4*x ppm. Repeat the proces to check and adjusta accuracy

Dareblix, 2016-04-19:
It's hard to be dumb, so it's hard to be me, i don't get it :/ the rod i use it is 1/4" 20 for me it's really hard to see the steps if need to advance one mm, but to see this i need to move the motors manually with current and no with the drivers isn't?

electrokean, 2016-04-19:
OK, so your threaded rod is 20 TPI - thus you need to rotate it 20 times to move one inch.  Now you need to work out how many steps per rotation, which will be dependant on your motor and driver.

The Easy Driver defaults to 8 step microstepping mode. So if your motor is 200 full steps per revolution, you would get 1600 steps/rev using EasyDriver.

Putting these two together, 1600 steps = 1/20th inch = 1.27mm
That works out to approx. 1260 steps per mm (1259.84252)

Can you try that?
You may have to adjust if your driver isn't configured for the default microstepping mode.

Dareblix, 2016-04-19:
Elecktrokean that's the number it gaves me when i use the steps per mm calculator http://prusaprinters.org/calculator/ 
well close i've used another calculator but the range it's close something like 1250-1300, i didn't went to use this because that calculator it is for 3d printers and i didn't knew if i was placing the right values so.. it sohuld be someting like this? 
0= 1260 x step, mm and same for y and x axis isn't?
Also someone please tell me if catia it's a good software to make my g codes

Dareblix, 2016-04-19:
Also i'm not using the 5v alimentationof the drivers , if i use it that'll help?

electrokean, 2016-04-19:
Try it and see - we don't know enough about your mechanical set up until you give us more information on what happens when testing that setting.  Yes, you should set that for all axis (X/Y/Z) that use 1/4"-20 driven directly from your stepper motors.

Regarding Catia, that is a pretty unusual choice unless you use it already.  The only experience I've had with it was on IBM UNIX and 20+ years ago.  I believe it does have CAM functionality, but I have no idea if it has a G code post processor/generator as standard that will work with Grbl.

electrokean, 2016-04-19:
"5v alimentation of the drivers" - I don't know what that means.  There is a 5V output on the EasyDriver, but you don't need to connect that.  A standard Arduino running Grbl has its own 5V power.

Dareblix, 2016-04-19:
I'll try it definitely tonight, what software to design and to create g codes do you recommend? something i can use with grbl

electrokean, 2016-04-21:
In the past I've used CamBam on Windows for simple CAD/CAM work and g-code generation, although I mostly use it with Mach3 rather than grbl.  With grbl I use my own g-code generator and sender, because I'm typically using it for robot automation rather than machining.

For part design and machining I am just starting to use Autodesk Fusion 360 now, which has been mentioned elsewhere here as working well with grbl.

If you have a good reason to use Catia, go ahead - but you probably wont find many others who can help you if you have issues.

Dareblix, 2016-04-21:
If a create g code with solidworks and masterckam would it work with grbl? I was wondering that

electrokean, 2016-04-21:
MasterCAM should be OK.  Solidworks is 3D modelling CAD software and can't generate g-code without add-on CAM software like SolidCAM or MasterCAM.  There are many other factors you should consider in choosing CAD/CAM software - like familiarity, pricing, training, support, etc.  Pretty much all CAM software will work with grbl, you may just need to use a g-code sender that post-processes any incompatible commands.

This is now quite off topic.  This site is for discussion of grbl issues, not what CAD/CAM software might work for your particular situation.  Try something and see if it works for you, and if not ask for help in the appropriate forum for that software - unless you hit a grbl specific issue.

Have you fixed your grbl configuration problem?  If so, please close this issue.

jahnj0584, 2016-04-22:
I use solidworks and SolidCAM without an issue. Friend uses cambam and also works with UGS. This thread was resolved with my answer that no one seems to have read.

Dareblix, 2016-04-22:
Elektrokean it worked!! Now i gotta work with a little more of precision, i mean like one digit more (1.50mm for example) but well this is ok for now, thanks to everyone here and sorry for be an asshole and talk about another topic, you are great guys

closed 2016-04-22 by Dareblix