grbl/grbl #966: How to draw a perfect circle?

sceadu50ys, 2016-04-19:
Hi, I'm newbie so sorry for immature question and poor English.

My laser cutter bought from Aliexpress, like this and I rewrote a 0.9j FW. 
http://goo.gl/yf3pFi

I wrote a 5cm diameter circle with Inkscape and export a G-code  with laser engraver extension.

The output command is...

F2000
G90
G21
G0 X74.8256 Y50.6748

M03 S255
G1F1000.000000
G03 X67.5646 Y33.145 I-24.7909 J0.
G03 X50.0348 Y25.8839 I-17.5298 J17.5298
G03 X32.5049 Y33.145 I-0. J24.7909
G03 X25.2439 Y50.6748 I17.5298 J17.5298
G03 X27.131 Y60.1619 I24.7909 J0.
G03 X32.5049 Y68.2046 I22.9038 J-9.4871
G03 X40.5477 Y73.5786 I17.5298 J-17.5298
G03 X50.0348 Y75.4657 I9.4871 J-22.9038
G03 X67.5646 Y68.2046 I-0. J-24.7909
G03 X74.8256 Y50.6748 I-17.5298 J-17.5298
G1  X74.8256 Y50.6748
M05 S0

F2000

G0 X0.000 Y0.000
M05
M02

And the result is above. it's a polygon :-(

[image]

Why am I unable to draw a circle?

[image]

Thanks,

chamnit, 2016-04-19:
@sceadu50ys : Set your $12 arc tolerance setting to something that is not zero. It should be about 0.02mm.

sceadu50ys, 2016-04-19:
Thank you for your quick reply, chamnit.

I changed $12 to 0.002(default), 0.02, 0.2...

And the result is...
[image]

How can I do next?

chamnit, 2016-04-19:
@sceadu50ys : Try a different g-code streamer like UGS or bCNC. It should work just fine, but I don't have a controller with me to test it.

sceadu50ys, 2016-04-19:
Thanks, I tested UGS, $12=0.002.

The result is...

[image]

Is this impossible for me? :-(

pth3k, 2016-04-19:
could you do a $$ & send us the output?

chamnit, 2016-04-19:
@sceadu50ys : Very strange. Are you using the version of Grbl that is posted here at this website?

Norwegian-Gadgetman, 2016-04-19:
Weird... 
Does GRBL return any errors when running this code?

I notice that some of the commands have a '.', but no digits after it.  Actually, there are 4 lines with that, and there are 4 long straight lines in the pictures, also.

sceadu50ys, 2016-04-19:
Hi, chamnit.

I downloaded this file : http://bit.ly/1EiviDk

Is it OK?

Hi, Norwegian-Gadgetman,

I used another extention, "J Tech photonics laser tool(Generate Laser Gcode)" it out put below

M05 S0

G90
G21
G1 F3000
G1  X51.0512 Y26.5766
G4 P0 
M03 S200
G4 P0
G1 F1000.000000
G2 X49.1683 Y17.1109 I-24.7353 J-0.
G2 X43.8064 Y9.0862 I-22.8524 J9.4658
G2 X35.7817 Y3.7242 I-17.4905 J17.4905
G2 X26.3159 Y1.8414 I-9.4658 J22.8524
G2 X16.8502 Y3.7242 I-0. J24.7353
G2 X8.8255 Y9.0862 I9.4658 J22.8524
G2 X3.4635 Y17.1109 I17.4905 J17.4905
G2 X1.5807 Y26.5766 I22.8524 J9.4658
G2 X8.8255 Y44.0671 I24.7353 J-0.
G2 X26.3159 Y51.3119 I17.4905 J-17.4905
G2 X35.7817 Y49.429 I0. J-24.7353
G2 X43.8064 Y44.0671 I-9.4658 J-22.8524
G2 X49.1683 Y36.0424 I-17.4905 J-17.4905
G2 X51.0512 Y26.5766 I-22.8524 J-9.4658
G1  X51.0512 Y26.5766
G4 P0 
M05 S0
G1 F3000
G1 X0 Y0
M18

and the result is...

[image]

Then, I re-write my arduino with 0.9i according to chamnit instructions. And send command above with UGS... and f...f...f...finally! WOW! 

[image]

I want to express my appreciation for your help.　Thanks!

chamnit, 2016-04-19:
@sceadu50ys : Ok thanks. Depending on the v0.9i build date you used, there was an arc bug fix around March 2015. Not sure if your problem is related to that, but at least it's now working right.

closed 2016-04-19 by chamnit

cheton, 2016-04-19:
Grbl Controller 3.6.1 had a bug that may draw G3 arcs in the wrong direction with its visualizer. I'd recommend you using other G-code senders to send G-code files.

sceadu50ys, 2016-04-19:
Thanks, I'll use UGS :-)