grbl/grbl #967: I want to change that code

atakanzey, 2016-04-19:
Hi, I am doing tinkernut's mini Arduino CNC. but I want to use a servo motor for z axis. So I have to change the code .How i can do it .I can not convert that hexadecimal code into text code.Please Help Me thank you.

electrokean, 2016-04-21:
All the source code for grbl you need is right here on github under the Code tab
https://github.com/grbl/grbl/tree/master/grbl

atakanzey, 2016-04-21:
Ok but, How I can change the .hex code. Or which code will be converted into .hex file.

cheton, 2016-04-21:
Just check out the wiki:
https://github.com/grbl/grbl/wiki/Compiling-Grbl#via-the-arduino-ide-all-platforms
https://github.com/grbl/grbl/wiki/Flashing-Grbl-to-an-Arduino

You can change the source code in Arduino IDE and upload Grbl to your Arduino.

cheton, 2016-04-21:
This might help you: https://github.com/robottini/grbl-servo

closed 2016-09-03 by chamnit