grbl/grbl #968: Error: Invalid statement (for G1 F250)

lachap1234, 2016-04-20:
Hey a got this problem when I create a .nc file with inkscape and I lunch it in grbl. Is there a way to fix that? Or maybe an other program instead of inkscape?

Norwegian-Gadgetman, 2016-04-20:
Just the one line?
Could you post the code up until the error appears?

lachap1234, 2016-04-20:
https://gyazo.com/d1274d450c82d488550c92d6d6bdf1bb

look at the third last line

maaadbob, 2016-04-20:
So the command is move at feed rate (250) to a position XY, but where are your XY co-ords ?
Correct usage would be:
  G1 X100 Y125 F250

Bob.

maaadbob, 2016-04-20:
>...Or maybe an other program instead of inkscape?

I'm looking at Fusion360; a combined modeller, simulation and CAM program.
Have not tried its output with GRBL, but GRBL follows the CNC standard and I'm sure 
Autodesk do the same.

Bob.

chamnit, 2016-04-20:
@lachap1234 : I recall making the `G1 Fxxx` command valid for Grbl a while ago. It also looks like you are using a very old version of Grbl, based on the settings printout in your link. I suggest that you update Grbl to the version posted here on this site.

closed 2016-04-20 by chamnit

lachap1234, 2016-04-20:
Where is the update version on the website? Do you have the link to the good direction?

tklus, 2016-04-20:
Fusion 360 works great with grbl. There is a grbl post and it works well. I have been using fusion 360 for over a year now. And it's free to hobbyists!

I'm looking at Fusion360; a combined modeller, simulation and CAM program.
Have not tried its output with GRBL, but GRBL follows the CNC standard and I'm sure
Autodesk do the same.

Bob.

-