grbl/grbl #969: Optical limit switch being triggered

1KHz, 2016-04-21:
I have built a cnc router. I am using an Uno, v0.9 grbl, I decided to use optical limit switches because I liked not having a switch to crash into, along with a little more accuracy. The switches I got are, output HIGH. 5V supply from Uno. So I inverted the limit switch input in grbl. Now, keep in mind I didn't take much caution in wiring shielding, the limit wires are inline with the motor wires. Im sure that is the majority of the problem. 

I am having a problem with "HARD LIMIT", being triggered for no reason, in the middle of a job. I have a Rotozip as a spindle, at the moment. I can get it to trigger "HL" almost ever time I turn it on. I can wrap my hands around the motor and that seems to stop the "HL" trigger, from turning the rotozip on.

Is there a way to change the sensitivity of the input? Maybe changing scaling in the code? The more I think about it, maybe a small capacitor will control spikes/noise.?

I purchased a 1.5kw china spindle for it. So I will be rewiring and doing some shielding at that point.

splitn2, 2016-04-21:
best and only reliable solution is to tidy up your wiring and add RC circuit suppression, trying to do this with code is not going to be effective in my view. Shielding and earthing practices in CNC are super important, in my electronics radio tech training we learnt about amplifiers and the most important lesson was tie all earths back to one point, earth cable shields at one end only.... those 2 rules go a long way towards a good outcome, troubleshooting this stuff only makes cnc frustratign when you are already learning a whole lot of new stuff don;t make it hard for yourself eh. cheers

1KHz, 2016-04-23:
Great advice! What would you recommend for shielding? I have read, braided copper sleeves would be the best.

splitn2, 2016-04-24:
@1KHz I suggest use use some shielded twisted pair cable its cheap, the stuff people wire buildings with for ethernet networking. Connect all of the various switch cable shield wires together at the controller end and do not connect the shield wires to anything at the switch end. So all shield wires come back to one common point. Call that point your FRAME GROUND and it can be coupled to the GND or DC GND (0V) with a small capacitor maybe 0.01uF.

1KHz, 2016-05-23:
So I got everything all rewired, and the new spindle mounted. I used your advise, splitn2(Much thanks BTW). I also put a 10K resistor, and a 0.1uF ceramic capacitor, on each axis limit pin (XYZ). Along with a 0.1uF ceramic capacitor inline with the wire shield drain wire, and 0V GRD.

After all of that, everything seems to be perfect. I wasn't able to get the machine to false trigger a hard limit. I also got some info from BenHarper, over on Instructables.

splitn2, 2016-05-23:
excellent @1KHz glad to hear you're winning!

closed 2016-09-03 by chamnit