grbl/grbl #970: Issues with T5 M6 commands?

jahnj0584, 2016-04-22:
Im not sure if my CAM is suddenly creating this (I did not update it), but UGS does not play ball with any files that use the t5 m6 commands. Is this something I have to change in my post processing settings, or is it just the newer versions of UGS being confused?

Thanks will.

jaketeater, 2016-04-25:
T5 M6 means do a tool change (M6) and pick up a tool from pocket five (T5).  M6 is not on the list of GRBL's supported g-codes.  (See the bottom of this page for the list [https://github.com/grbl/grbl](https://github.com/grbl/grbl) )

chamnit, 2016-04-25:
@jahnj0584 : @jaketeater is right. M6 is not supported by Grbl. Usually GUIs will remove that line of g-code automatically. I would either remove that line manually or ask the UGS developers about it.

closed 2016-04-25 by chamnit

jahnj0584, 2016-04-25:
I think i got it. I had my early bits set to manual tool change, and stopped checking that with these newer bits which I guess I DO have to do.

chamnit, 2016-04-25:
@jahnj0584 : FYI, some GUIs, namely bCNC, do support M6 tool changes, where they look for that command and insert a customizable tool change procedure.