grbl/grbl #972: grbl without Gcode interpreter, someone interested ?

cri-s, 2016-04-25:
Are others interested in GRBL  code , 11 Axis max (U-E) that 
have gcode and motion_control on PC side and on uC replaced by fifo module.
I want doing that for testing advanced spline or ramping algorithm.
The api should allow interface with planner, stepper and should having some trace buffers.

chamnit, 2016-04-25:
@cri-s : This type of model isn't all that different from LinuxCNC or even Mach3, because you are adding the requirement (and dependencies) of a PC to make it work. It's always good to have options, if you go through with this idea, but I don't see much upside.

closed 2016-04-25 by chamnit

cri-s, 2016-04-25:
Adding i2c memory it can be stored inside flash. The advantage and why I
want doing it is delta motion planning using splines on acceleration and
path.

chamnit, 2016-04-25:
@cri-s : If you store data in external memory, there isn't a reason why a micro-processor couldn't pre-process relevant planning data itself. Either way, I'd be interested in seeing a spline-based planner. They have lots of theoretical advantages. Whether those advantages are truly useful or significant for the computational costs, that remains to be seen.

langwadt, 2016-04-25:
I guess it could be a path for using it as a controller/IO for linux CNC, don't see many PCs with parallel ports anymore

cri-s, 2016-04-26:
Yes, that is a good argument. Need to check it out what is required.
Actual max rate is 66khz or 20.833 rotation / second at 16
microstepping and 200 steps/revolution for 6 axis grbl code. I don't
have made measure for 11 Axis code.
Without big optimisation, 42-45khz is reasonable, with optimisation
140khz should be doable.

cri-s, 2016-05-05:
question to   Sonny Jeon:

Is there interest for higher speed arduino (nano) code, or in sight of
ARM target, it's not worth
investing you time for that ?
stepper.c replay only step and dir commands inside interrupts from
queue that is feeded from
planner.c. Pause and stop ramp down is handled lowering the interrupt
cycle time.

#define QUEUE_SZ   32
#define QUEUE_MSK  ((1<<QUEUE_SZ)-1)
volatile uint16_t queue[QUEUE_SZ];

void inline queue_full() {
        return queue[idx_push];
}
enum { Q_STEP, Q_DIR=2, Q_END };
/*
        push value, ddddddff
        ff:
           00 = step
           10 = dir
           11 = end -- dddddd must be zero
        return false if queue is full
*/
char queue_push(val) { char valid=0;
        static uint8_t last;
        static uint16_t value;
        if((last&3)==0) // step
          if(val==last) {
                if(++value&0x200) { // overflow
                        queue[idx_push++&QUEUE_MSK]=value|0x1ff;
                        value^=0x200; return;
                }
          } else {
                        queue[idx_push++&QUEUE_MSK]=value|0x200;
          }
          value=val<<8; last=val;
        if((val&3)!=0) {
                        queue[idx_push++&QUEUE_MSK]=value^0x300;
        }
        return !queue[idx_push];
}
/*
        push value, ddddddff
        ff:
           10 = step
           01 = dir
           00 = end
*/
uint8_t queue_pop() {
        uint16_t value = queue[idx_pop&=QUEUE_MSK];
        if (value)
          if (value&0x200)              // step
                if(queue[idx_pop]--&0x1ff); else goto drop;
          else  drop: queue[idx++]=0;   // dir
        return (value>>8);
}

(pseudo)code on stepper.c file: // 6 stepper support, different pinout
as official grbl !
        if(i=queue_pop()) {
                if(i&2) { // STEP D2-D7
                        PORTD=i^settings.step_invert_mask;
                } else  { // DIR  A0-A3,D8,D12
                        i>>=2;
                        PORTC=PORTC&~0xf|i&0xf; // C0-C3 -- A0-A3
                        i&=~1;  if(i&0x20) i++; // move bit5 to bit0
                        PORTB^=PORTB&~0x11|i&0x11;// B4,B0 -- D12,D8
        } else  {         // nothing to do
        }

2016-04-26 8:44 GMT, Cri S <phone.cri@gmail.com>: