grbl/grbl #973: Which cad software do you recomend for grbl?

Dareblix, 2016-04-25:
Hello everyone, i was wondering what is the best choice for create parts with grbl, i don´t have to much experience with these so that´s why í`m asking you, i started with catia but i don`t know if those g codes are usefull with grbl, but please tell me what do i need, and if catia it`s ok for machining with grbl, Thanks for read!

tmpvar, 2016-04-25:
I think my preferred way of doing things these days is using http://onshape.com (free) for the CAM side, then export and use http://www.autodesk.com/products/fusion-360/overview (free for hobby use) for what is basically HSMExpress.  I'm not a huge fan of Fusion360's CAD interface.

Hopefully that helps, let us know what you try!

Dareblix, 2016-04-29:
Well, solidworks it´s very easy to use, i just downloaded fusion 360 and it`s pretty nice, i`ll try to export my designs from solidworks to fusion 360, Hope it works for grbl!

blinkenlight, 2016-05-06:
Depending on how simple your designs are, you might want to take a look at [GCAM SE](https://github.com/blinkenlight/GCAM/releases) - it's a really simple 2.5D CAD/CAM and it's what I use with grbl (not exactly surprising since I'm the one maintaining it at the moment). Feel free to take a look at the [Wiki](https://github.com/blinkenlight/GCAM/wiki) for details or screenshots if you're interested.

Dareblix, 2016-05-06:
If i use solidworks and i make the gcode with solidcam would it work with grbl?

jahnj0584, 2016-05-06:
yes, just set grbl as your post processor.

Dareblix, 2016-05-07:
What do you mean?

jahnj0584, 2016-05-07:
you do your CAM in solidcam them export it to grbl.........

Dareblix, 2016-05-07:
I see... Thanks for your time another question mate if i want to make a cube for example 1" x1" measures do i have to draw that cube in 3d with that measures? Or i gotta draw the square and set 1" of depth? I don't know if you know what i mean...

X3msnake, 2016-05-07:
If you are doing full 3D paths with your CAM software you do it the first
way else you do it the secound way.

For flat shapes like that it better and faster to do it the secound way
called 2.5D

The full 3d mode usually implies using 2 operations. One of roughing and a
finish pass

Dareblix, 2016-05-17:
Ok i made my desicion i'll use solidworks+mastercam it will work with grbl isn't it?

tklus, 2016-05-17:
MasterCam will work fine if you have a post for GRBL.

I also recommend fusion 360 for CAM programming. It is free for hobbyists to use and works quite well. There is also a post in fusion 360 for GRBL.

Also, unless you have access to MasterCam, it is quite expensive!

Dareblix, 2016-05-17:
Thanks for answer, what do you mean with post? Well i just cracked mastercam xD

chamnit, 2016-05-17:
@Dareblix : If you don't know what a post (post-processor) is, then you probably shouldn't be using Mastercam. It's for professional machinists and has a fairly steep learning curve/not very intuitive.

closed 2016-05-17 by chamnit

tklus, 2016-05-17:
I agree! I use MasterCam at work and have years of experience working with it. Fusion 360 is much easier to use and much more intuitive. and it's legal to use [😉]

FYI, a post is a set of instructions that the CAM software uses to generate G-code for specific machines. it takes the CAM operations and interprets it into something that the machine can read.

Dareblix, 2016-05-18:
You're right i need more experience, what's the best choice then? Learn fusión 360 first? 
I'd like to export solidworks draws to fusión 360 
Any link where i can learn how to machining with fusión 360 using grbl?

johnamptech, 2018-08-15:
I have a question in two parts,i want to make a les paul style guitar with an arduino ,grbl shield v3(or later) based 3 axis cnc,this particular guitars body has compound curves and deep neck pocketing i think this is 2.5D,i can buy g-code files with body and neck design files,what would be the easiest route to do this,from a cad/cam perspective,at present using inkscape i can only get the grbl controller to work with .nc files.and it seems rather convoluted to go from say a jpeg to .nc files to draw.
I also use Tina design suite for circuit simulation and pcb design,so i have used cad,though i've never exported gerber to make the pcbs,which is another thing to do.
I need to upgrade to win10 as the vista OS seems to be lacking in some areas such as new software compatibility .
I'm learning on a cd-rom 3 axis ,which will be using end stops and a laser,it has a pen and i'll be fitting a dc motor to simulate a spindle ,basically a small version 3 axis to get to grips with this learning curve.
I have 5 tbm6600s, 3 nema 17s ,3 acme T8 lead screws to make a small machine before i go to a unit big enough to make a neck and a body.
The intermediate unit will need to work at least a 17" x 2" brass front panels i want to etch/engrave.
I'd go for a break out box and mach3 ,but most don't support usb,so that's out,at least for now.
Any solutions to getting g-code to grbl without crashing in a relatively easy manner would help in a big way.
Thanks.

VhbMo, 2020-01-02:
Hi All,
I want to draw my schem in Altium Designer and then can prepare both the Cam file and DXF. The question is which software is better? I read about GCAM, is it the correct one for G-code or it is better to use another software?
Thanks

blinkenlight, 2020-01-03:
I assume you intend to use CNC isolation milling to produce PCBs designed in Altium. GCAM can help you with that, as it can take Gerber output from other software and create g-code to produce the PCB. However, it also has some limitations to keep in mind if you intend to go that route: 
- it cannot handle DXF files, so if you also want it to mill the PCB outline you will have to either convert that to SVG (which GCAM can read) or you will have to (re)draw the PCB outline in GCAM manually.
- it cannot handle filled zones, so if you have any of those, it's not going to work for you.
- due to the previous point, it also cannot handle "rounded rectangle" pads that have recently become popular (which seem to get implemented as tiny filled zones in Gerbers), so if Altium is using those, you'll have a problem (although this might not be limited to GCAM alone).

There is other other software for isolation milling you might want to take a look at, some more recent than others, such as [Line Grinder](https://www.ofitselfso.com/LineGrinder/LineGrinder.php) or [FlatCAM](http://flatcam.org/). Good luck!

MeJasonT, 2020-01-03:
I have used both Proteus and Altium in the past but never used the gerber output although i have sent files to third parties for production.
The cost of buying copper clad board and milling out your design will not be very cost effective.
Getting your boards flat and milling them to any sort of accuracy will not be an easy task.
For that reason i now use a third party company myself
https://jlcpcb.com/quote#/
take a look at them, they are dirt cheap for 10 boards and their finish is superb.  I have used a commercial manufacturer in the UK in the past and their quality was absolute pants compared to jlcpcb

MeJasonT, 2020-01-03:
Fusion 360 does my tits in for two reasons
i have always sketched my designs using basic drafting tools in Autocad, creating regions and then extruding my solid parts - i have not quite figured out how to do it in fusion yet.  slowly going through Lars Christians turtorials on youtube (he works for autodesk).
second they have now removed the file translators from the free/student version.   Great thanks guys - wait until you grbl guys try exporting files from Fusions cloud to your PC, its going to be fun.
I tend to obtain parts from grabcad as its easire to get bearings, gears etc from guys who have already drawn them up.  for some reason being an American site nearly all the files are in solidworks file formats. (Dassault being European and Autodesk being American).  Fusion 360 is going to stop importing and translating these files, also exporting files from the cloud will be restricted. I knew it wouldnt be long before autodesk started to take the toys away.  More fool us for trusting in there apparently free application.