grbl/grbl #974: Homing issues

Hineswoodworks, 2016-04-26:
Well from what I've read I'm pretty certain I have noise issues. I have my switches wired NO and I was relying on the internal pullups but now after re routing some wires I'm getting random triggers during the homing cycle. I'm using shielded wire but I am running it past a stepper and alongside the stepper wires for all 3. Will adding external pullups help or will I need to do something more substantial?

chamnit, 2016-04-26:
@Hineswoodworks : External pull-ups will do little to help. But, adding an RC circuit (or low-pass filter) to all of your limit switch pins will help tremendously. Also make sure to ground your shielded cables to a single common ground point (to prevent ground loops).

Hineswoodworks, 2016-04-26:
OK here come noob coming threw. To ground my cables I can run the ground wire in my sheilded cables to the ground on the Arduino or ground it to a power supply?

Hineswoodworks, 2016-04-26:
@chamnit Thanks that worked! Nothing worse then having a machine crash and fail the first time you show a perspective buyer your machine.

tklus, 2016-04-26:
Did you ground your shielding to the Arduino?

@chamnit
What is an RC circuit?

Thanks

Hineswoodworks, 2016-04-27:
@tklus I grounded to the ground pin on my input on my power supply ( so grounding to the wall ). Don't know if that was right but seemed to work. 

I had to look it up too but an RC Curcit is a Resistor Capacitor Curcit. I used this threadhttp://www.shapeoko.com/forum/viewtopic.php?f=4&t=1290&p=31052 to wire mine. The info I found helpful was the large quoted section after the diagram. From what I can tell the resistor (internal on the adruino) keeps the pin held high and the capasitor keeps it held at a constant high even when there is interference causing it to go to low. Don't know if that helps / is right.

electrokean, 2016-04-27:
Here is a good post on adding a capacitor to the limit switches, showing how it helps reduce noise
http://wot.lv/combating-endstop-noise-on-a-reprap.html
Note that the series resistor isn't really required - the key things is the combination of pull up resistor to 5V and the capacitor to GND which combine to filter high frequency noise.

electrokean, 2016-04-27:
All grounds should ideally come from one central point (the power supply) and connect out in a star fashion to each cable/device.  This avoids ground loops, another source of problems.

Destes1, 2016-07-27:
Hello all. New to the site and cnc, but beginning to understand things somewhat better as I read and play with the 3 axis router I have built. I'm running grbl through a g shield on the arduino uno with Y cloned to A,   for the fourth motor. Everything has been going smoothly until I installed my limit switches. They NO lever switches on both ends of each axis, wired in parallel, with small capacitors in circuit for each axis limit switch. They all go to ground properly when closed. (well they all have continuity when closed at least.) My issue is when I attempted to run the first homing cycle the z axis touched the switch and pulled off,then x and y both moved to their switches and touched off and pulled back. Then GRBL reported "Homing Fail". I thought that after the last y pull off homing was complete, so I'm not sure what failed. Any ideas?

jahnj0584, 2016-07-27:
It's possible it thinks it hit a hard limit and not the home limit switch.
You may have to change the pin layout in software which I believe is $6 or
something.

Destes1, 2016-07-27:
I don't have hard limits enabled yet, so is that still possible? $6 is probe invert according to the chart. I'm running GRBL.o9

jahnj0584, 2016-07-27:
Then my memory is wrong :P

I assume X home is close to you, Y home is left, and Z home is as far away
from the floor as possible? I had to apply an invert mask to my homing
procedure.

Destes1, 2016-07-27:
It homed all the way to the back right corner, essentially at the end of the positive space. I have read that is the proper home position for the Machine. I would probably set my Work starting point to my lower left, but as I understand it (I think) it is common for the machine position to be toward the back right.

Destes1, 2016-07-27:
And yes, the Z is all the way up.

Hineswoodworks, 2016-07-27:
I've got mine homing to the the front left cus I fint it a bit easier of a layout. Your machine wouldn't do the step off if it was hitting a hard strop.

jahnj0584, 2016-07-27:
The top right is for milling machines where the bed is moving, so it
results in the bed being out of the gantry and ready to be loaded. A simple
edit in the config file fixes that.

Destes1, 2016-07-27:
That's what has got me confused. It's as though homing went thru all the steps successfully and then reports as a fail. Maybe I'll try a reset and start homing fresh and see if it changes. Thanks for your input. I have to leave the forum now, but will def be back.

chamnit, 2016-07-27:
@Destes1 : A "Homing Fail" occurs when: The limit switch is read as triggered before starting the homing cycle. The limit switch is still engaged when the pull-off motion fails to move far enough away from the switch (increase pull-off to fix). The homing cycle fails to find the limit switch within 1.5x the axis travel. It'll automatically stop and not go on forever.

Destes1, 2016-07-27:
OK. I am going to home again and check to make sure all the switches are clearing. The Z obviously is ok or it wouldn't move on to the x and y. And I suppose the X is ok, because the Y axis is much longer and it is still moving toward Y after the X has pulled off. So it has to maybe be the Y that is the problem. I'll try again and report back.

Destes1, 2016-07-27:
Bythe way, what is the report back from grbl when Homing has finished successfully?

jahnj0584, 2016-07-27:
Both axis will eventually click, and they then debounce when they are both
at their extreme points. Try increasing your debounce distance to 2mm
otherwise the switch not be able to toggle.

After you home it just resets machine position to 0,0,0 and removes the
alarm.

Destes1, 2016-07-27:
OK, tried again and set the homing pull off to 4 mm. I thewn jogged the Y axis back toward the middle of the rail and ran Homing again. Zied was fine.When it moved on to X it clicked the switch and stayed there and moved on to the Y switch. It clicked the Y switch and then both X and Y pulled off together. It sat there for about 2 seconds and then the dreaded red alert,,, Homing Fail

Hineswoodworks, 2016-07-27:
Try adjusting you debounce not just the step off

Destes1, 2016-07-27:
debounce was set to 50. What is recommended?

Destes1, 2016-07-28:
OK it's round two. I  set the  H debounce to 100, the pull offback to 2,  and by-passed the .47mfd caps I had in the limit switch circuits and tried homing again. This time the Z axis did not reach the switch before stopping and pulling off the 2mm, then the x and Y axis both reached their switches and pulle off 2mm. No Alarm, just an OK. Problem is now, the z axis is almost 40 mm away from the switch!. Tried $H again, same thing. Any ideas?

jahnj0584, 2016-07-28:
is it possible you have electrical noise on your Z axis, causing a false
trigger?

Destes1, 2016-07-28:
Also, after the homing, the machine position reads X -2, Y--1.98, and z -2 , even though it's way off the switch.

Destes1, 2016-07-28:
That is my guess. I tried re-connecting the cap to just the Z leads, but then the Z didn;t even move. I have Z limit to the Step enable pin as advised for GRBL.09 Not sure if there is a factory resistor on the Arduino for the pin as there are for the limit pins. I've read elsewhere that the caps are more effective than resistors, so went with the cap. Is it possible I need a resistor instead?

jaketeater, 2016-07-28:
Check your $100, $101 and $102 settings.  It appears that the number of pulses per MM is not correct.  If one of these settings is too high, it would cause it to travel too far.

Hineswoodworks, 2016-07-28:
I had similar issue and resistors did nothing but the caps did. I have my step off at 20mm. Try higher then 2mm. May not be enough to deactivate your switch. Where are you caps in the circuit?

Destes1, 2016-07-28:
I have the shielded limit switch wires coming into a main panel box to a copper grounding bus that grabs the shield wrap then the two leads go to a small perf board with a row of pins for ground, and three .47 caps between the ground strip and three sets of dual + pins. The + leads go to their respective pins followed by jumpers to the G shield. Double checked the 100-102 settings and they are correct. is it possible the caps are too small or large?

jaketeater, 2016-07-28:
Can the presence or absence of capacitors on the limit switch wiring have any effect on jogging length (besides interrupting a jog)?

Can you jog the machine 2 mm and measure the actual travel?  For example, if you have a 2mm pull off, your machine should be at X -2 after homing.  Enter:
`G0 X-4`
to travel 2mm.  Verify the distance.

closed 2016-09-03 by chamnit