grbl/grbl #975: Motion completion response without using G4 P0 or Idle state checks

milesingrams, 2016-04-26:
Firstly I want to say that grbl has been an incredible tool for my project, and thank you for all the work you have put into it.

I wanted to ask about the possibility of providing the option for a response after each motion has been completed. Perhaps an 'ok' when a move has been read and a 'done' when a move has finished. While I know this is possible using the 'G4 P0' command or by polling for an 'Idle' state, both of these options have major drawbacks and feel like hacks. Foremost is the issue that neither of these methods cooperate well with the motion planner and don't allow for smooth motion.

I don't know much about the intricacies of how grbl works, so I don't know how hard it would be to implement, but I think it could be a very useful feature for anyone building a GUI that shows the current progress as well as for many other applications that require some function that depends on knowledge of the current progress.

Thanks!

- Miles

chamnit, 2016-04-26:
@milesingrams : Yes I would agree that the 'G4 P0' and 'Idle' polling isn't the best solution, but it's mainly a product of keeping the flash footprint as small as possible on the 328p AVR. I suppose knowing exactly when a command has completed would be somewhat useful for GUIs, but there are some problems.

For one, you'll need to have Grbl retain some knowledge of what it's executing and then report back to the GUI what that thing was it finished. To keep things small and compact, the g-code line data is thrown out after initial parsing of the g-code command. 

One solution has been to use line numbers to indicate to the GUI where Grbl is in execution. You enable it through the config.h file and Grbl will then report the line number that its currently executing in every status report. It's the job of the GUI to make sure to send a unique line number ('Nxxx') with every line and know which number goes to which g-code line it sent. In this way, you can track exactly where grbl is and what its doing.

milesingrams, 2016-04-26:
Thanks for the quick response Sonny!

The line numbers solution would be an excellent addition. To clarify, I think this info is more useful than just to GUIs and visualization of progress. For instance I would use the info to keep track of the current progress so that if a hard limit or soft limit is tripped the machine can continue from the last command it was executing after the cause of the problem has been resolved and grbl has been reset and homed. With just "ok" feedback from the command buffer I have no clue where to resume from.

chamnit, 2016-04-26:
@milesingrams : Line numbers is an existing solution and can be enabled through config.h.

What you're alluding to is a program restart. It's not quite as trivial as you may think to restart a job safely. This is a dangerous thing to do, if you aren't very careful. G-code is a modal programming language. If you don't track and properly set all of the modes, coordinate offsets, and positions prior to the line that failed, all lines after will not execute as programmed. This is especially problematic if a GUI tries to parse the g-code itself and the controller has a slightly different parser (more common problem than you'd think). In addition, you have to ensure that a re-entry into the program path is done safely, which is dependent on where the tool is before the resume and how the machine is configured.

Cygnus-X3, 2016-05-07:
Hello, sorry for hijacking the thread.
I actually need the same kind of solution, i am building a device that use this firmware, it uses a probe to deliver some mechanical stimuli, and i need to know the exact moment the probe delivered it's stimuli.
i don't need the ok before the execution, i would need it only after, is there anyway to alter the firmware to do so?

cri-s, 2016-05-07:
And why you cannot use
G1x3f333 g4p0
This way the OK return after motion, not before.

chamnit, 2016-05-07:
Or use the G38.x probe commands? That's what they are for. Grbl will report the position immediately after it detect a trigger. And then stop. Latency (Grbl and USB) can be an issue if you need something else that is time critical, because an ok method may take tens of milliseconds by the time you receive the message.

closed 2016-09-03 by chamnit

erwzqsdsdsf, 2024-10-21:
> 
> > Hello, sorry for hijacking the thread.
> > I actually need the same kind of solution, i am building a device that use
> > this firmware, it uses a probe to deliver some mechanical stimuli, and i
> > need to know the exact moment the probe delivered it's stimuli.
> > i don't need the ok before the execution, i would need it only after, is
> > there anyway to alter the firmware to do so?
> > [#975 (comment)](https://github.com/grbl/grbl/issues/975#issuecomment-217652908)

you saved me！！！！！！！！！！！！！tongs of thanks to you