grbl/grbl #976: USB Serial port connection problems with arduino ...

ecncshop, 2016-04-27:
I have been using arduino UNO loaded with GRBL 0.9i firmware and UGS.
Everything was working fine. 

Recently I faced a problem in a laptop running windows 8.1

The serial port connection used to get disconnected frequently.
I was facing this problem the whole day. 
Gcode will run for sometime and suddenly get disconnected. ( I get port disconnected exception )
Actually I had a mouse also connected to another USB port.
In the evening, I unplugged the mouse from laptop.
After this : the problem stopped !!

In between I had tried DISABLING the USB selective suspend setting. 
That did not help.

Later I found another USB port related power setting : "Allow the computer to turn off this device. "
This setting can be accessed by going to : DeviceManager --> USBdevices --> USBrootHub --> properties-->Powermanagement. 
There r several USBrootHubs listed in the device manager. 
I am not sure which hubs property to be set.
I did not try this actually. 
Wondering anyone has any experience with this setting. 

Anyways : Iam curious why I faced this problem the whole day, is it a common problem faced by other users when using arduino with a laptop?

Iam still not sure about real cause of the problem: Was it mouse related ( why should it be ?), was it a virus,
Is it related to % charge left in laptops battery , or is it related to some USB power setting of windows ?!

There was no major power drain from arduino.
In fact the stepper drives were given an external 5V adaptor. ( with common ground to arduino )
So the step, direction pins ( X, Y, & Z ) in arduino were only sinking the current ( not sourcing ).

Another sub question : Is it recommended to use an external 5V to the drives or should I use arduinos 5V or should I just use Arduinos TTL signals itsef ?
Which is recommended. I would think external 5V would be better if one can handle the pains of wiring and accomodating it. 
But I have faced some problems in this also, which I will post separately.

electrokean, 2016-04-27:
I doubt it is a power management issue.  Are you running the laptop from battery?  Seems unusual with a CNC as you'd generally need mains power available for the motors.

From my experience, this is most likely a poor quality USB cable connecting the Arduino.  I've seen this a few times.  Try to find a cable with heavier conductors (not too thin a cable) and with a ferrite core (the bulge you see on good cables) at one or both ends.  The ferrite stops high frequency noise spikes entering the USB port (from your stepper drives) and causing a disconnect.  These ferrites are there for a reason (to pass EMI compliance).  Manufacturers would leave them out if they could as they add significant cost to the cable.

You can also buy snap-on ferrite cores to add to existing cables, but most people already have a random selection of USB cables and you may find one with a ferrite.  e.g. Samsung ship these types of cables with their LCD monitors.  Use Google image search on "usb cable ferrite" if you're unsure what I'm talking about.

Another possibility is if you're using a poor quality Arduino clone, with either poor USB noise immunity, or a crappy USB/serial interface.

Not sure about disconnecting of the mouse - could be the mouse cable acted as an antenna and picked up additional electrical noise.  If you need the mouse, try adding a snap-on ferrite core to its cable near the USB plug.  Or try another mouse.

If you have external 5V available for the drivers, then it makes sense to use that rather than the Arduino power, but it probably wouldn't make much difference in a simple set up.  Just keep you power cabling neat, and try keep all ground wires going back to a common star connected ground.

ecncshop, 2016-04-29:
Thanks for the detailed reply. 

My USB cable was without ferrite core. I will try cable with ferrite core. 

But still I dont understand how removing the mouse solved the problem. 

Actually my mouse USB cable was plugged on one side of laptop and arduino USB cable on opposite side. So dont think it would have had any antenna effect!

I still suspect this is related to USB port power settings.

If it is indeed so: several other users using laptops should have faced this problem. 

I found 1 or 2 discussions related to this :+1: 
https://plus.google.com/109806197474584466964/posts/PdXwZcr8yr7

https://github.com/grbl/grbl/issues/444

But both the discussions r inconclusive about this problem.

chamnit, 2016-04-29:
@ecncshop : This definitely sounds like a USB problem on your host computer, likely something driver related in Windows. I'm closing this thread, as it's not a specifically a Grbl-related problem, but feel free to post a solution if you find one.

closed 2016-04-29 by chamnit

soonermike, 2017-01-01:
Just a comment in case someone else is having the same issue.  Trying to run GRBL for the first time, it would not open the com port.  Trial and error.  I finally tried changing USB (arduino) to different connection on  PC and it fixed the problem, it looks like I didn't have enough power to drive it.  It may not be a big deal to some but to the rookies out there it was a major find.  Hope this might help someone else.