grbl/grbl #977: Arduino uno + Motor shield R3 + 42byghw609 stepper motor

Kingdombed, 2016-04-27:
Hello everyone !

I'm working on a project and I would like to know if an arduino uno with an arduino uno motor shield R3 can run the GRBL ? I'm able to upload GRBL in the arduino but i don't know if I can find a way to configure it. I'm using a 42byghw609 stepper motor .
This is my configuration:

[image]

It is possible to use GRBL with this config ? What config did i need to put to make it work ?
I've made some test and i can only make the motor turn a little bit before having an error : 

[image]

Thanks for your help ( sorry for my english, it's not my main language )

electrokean, 2016-04-27:
Unfortunately you can't use that motor shield with grbl.
Although it can be used to drive a stepper motor in theory, it actually is designed to operate two brushed DC motors, and thus requires two separate direction and PWM signals - rather than the step and direction signals that grbl generates.
It also uses the completely wrong pins for the motor signals - compare the pins allocations on these two pages -
https://www.arduino.cc/en/Main/ArduinoMotorShieldR3
https://github.com/grbl/grbl/wiki/Connecting-Grbl
You need a proper stepper driver board that takes step and direction signals - like the EasyDriver or BigEasyDriver.

electrokean, 2016-04-27:
Oh, and you appear only providing power via USB.  That wouldn't work for driving a stepper motor which will need more power (voltage & current) than USB can provide.

Kingdombed, 2016-04-27:
Ok 
I'm have a source a voltage to power it but yeah , i can't do more with the motor shield
Thank you, Have a nice day

closed 2016-04-27 by Kingdombed