grbl/grbl #978: using grbl with software

Encarnacao, 2016-04-27:
when u want to control grbl with other software u got to set up the serial ports on the software for what i understand , grbl uses baudrate 115200 parity none data bits 8 and stop bits 1 flow control i would assume flow control would be hardware or software hardware.
i was trying to make it work and actualy it works, sends the data to the arduino with grbl on it , but sends all gcode lines in one time, so machine goes to the end of the gcode and does the last lines and stops ,im  guessing is cause the buffer size to small for all the data to come in, 
i went to configuration of transmit section  and i have set the  handshake  timeout(seconds)for 10 and the delay before eatch line(ms) for  1000 
 now is working but im not sure if this timings is the correct ones to use i see also alot other settings that i didnt use CR/LF, omit lines, remove charaters, start triger, break after recive charaters, send and start feed, send at end of feed, wait for Xon, wait start button, replace tabs with spaces, remove white spaces, it also haves the recive set up with i didnt do nothing.
so i was hoping anyone that is also using arduino with grbl firmware could give a input in how to set up the correct setting

chamnit, 2016-04-27:
@Encarnacao : Please read the Interfacing with Grbl wiki page. It will be a lot more clear if you do this first.

closed 2016-04-27 by chamnit

Encarnacao, 2016-04-27:
i actualy have read it a couple times i kind of get the idea, im new into this and its a litle confusing how the all process real work in how to set up, i have look and look and so far i didnt find no exemples in how to realy set up the grbl arduino on the serial comunications part :(
the machine is actualy working normaly when i set the 115200 parity none data bits 8 and stop bits 1 flow  hardware,handshake timeout(seconds)set to 10 and the delay before eatch line(ms) set to 1000
just wanted to make sure that im on the right track and not do anything that would compromisse my machine...

chamnit, 2016-04-27:
@Encarnacao : There is a file called stream.py in the doc folder of the source code. Please review that to see how it works.

AlexHolden, 2016-04-28:
@Encarnacao I think what you may be missing is that the sender software has to wait for grbl to respond before sending the next line. Just uploading a file using a serial terminal program will cause an overflow because the Arduino doesn't support flow control.

Encarnacao, 2016-04-28:
Is there any way to set the software to wait the response from arduino grbl
to say ok,  on the handshake settings i have tryed with some values set to
1 is to fast sending lines eventualy stops, on 1000 machine slows down do
something stop do something stop eventualy stops,  with 10 seems to do
better,not sure whats the apropriate times  also i have read about the grbl
defenition Xon with is disable by default, do i have to make it able.?
sorry about my question im new to this so bare with me

>
> @Encarnacao I think what you may be missing is that the sender software
has to wait for grbl to respond before sending the next line. Just
uploading a file using a serial terminal program will cause an overflow
because the Arduino doesn't support flow control.

cheton, 2016-04-28:
Hi @Encarnacao,

Most G-code senders should have implemented a simple call-response streaming protocol, that will wait for Grbl <b>ok</b>/<b>error</b> response before sending next G-code command. For example:
```
> G0 X0 Y0 Z0
ok
> G1 X10 Y10 F250
ok
```

I don't know what G-code sender you're using, but you can use some sample scripts here and see how it works.
- https://github.com/grbl/grbl/blob/master/doc/script/simple_stream.py
- https://github.com/grbl/grbl/blob/master/doc/script/stream.py

chamnit, 2016-04-28:
Technically, Grbl can support software XON/XOFF flow control, but in my testing (a long time ago), it's not reliable because it depends on consistent timing. Every host serial XON/XOFF implementation is different (timeouts, buffer sizes, etc) and with added delays of USB-serial conversions, all of these can mess with timing enough to corrupt the data stream. In my experience, it's very finicky to get settings just right to where an XON/XOFF stream to work well and not corrupt data.

Encarnacao, 2016-04-28:
Sending in grbl software i got no issues, i was tring test with Cimco and
send directly from there the gcode comands, just strugling on the nc
comunication set up for the serial port comunication, transmition, reciving
sets