grbl/grbl #979: Z axis pin always high

PocketNC, 2016-04-28:
Hello,

I am trying to set up some hall sensors on my machine, for them to work, I need the limit pins to have the internal pullup on pins 9 10 & 12 turned off.  I am running grbl v0.9 on a uno R3 

I have messed with both the $5 setting and the #define DISABLE_LIMIT_PIN_PULL_UP in config.h
but am unable to get pin 12 / Limit Z Axis to turn off its internal pullup.  

Pins 10 & 9, (X / Y limits) change state with both settings, but pin 12 is not.  

Has anyone else had issues with this?

Thanks,
Matt

chamnit, 2016-04-28:
@PocketNC : That's strange. The limit pin configuration code is pretty straightforward. Just as a quick check, do you happen to have variable spindle disabled in config.h? If so, this moves the z-limit pin to pin 11. If not, then can you provide a $I printout to get the build date of your Grbl version? And point out any modifications to either the code or defaults? Thanks

PocketNC, 2016-04-28:
The issue was a bad 328p.  Switched it out and everything works now!

Thanks for the quick response!

closed 2016-04-28 by chamnit