grbl/grbl #980: $ command extensions

J-Dunn, 2016-04-28:
I have been using GRBL with L6474 driver boards ( Arduino pin-out ) , which require some additional set-up parameters since they are a lot more configurable than the typical STEP/DIR boards commonly used. 

The boards use classic STEP/DIR pins but the rest goes through a SPI daisychain including ENABLE / DISABLE.

There's a bunch of parameters like slew-rate etc that can be set but the main difference is programmable current setting; a hardware error FLAG and status byte.

I have added a few $ commands to the GRBL interpreter to set and interrogate the various parameters  and an interrupt to respond to error pin input. The most common error conditions are over-temperture warning, over-temperture error ( hard fault ) and supply undervoltage condition. 

I am going to be tidying this up a bit now and wondered whether there are any guidelines for extending GRBL's interpreter so that I can avoid clashes with future work any maybe provide something of more general use.  

As an example I have commands like the following to set chip parameters for each board:
$P<board_id>:<param_id>=<param_value>

Boards are becoming more configurable and this sort of situation will start becoming more common.  If there is not any defined way of extending the current code set, it may be good think about providing one. 

The primary requirement will obviously be to avoid clashes with any NIST GCODES and anything that may be tentatively in the making for GRBLs additional commands. 

Any suggestions?

chamnit, 2016-04-29:
@J-Dunn : I think this would be fine for the Grbl-Mega port I've been working on [here.](http://www.github.com/gnea/grbl-Mega/) I don't think cramming more and more stuff into the 328p version of Grbl makes that much sense if it doesn't have to be there. Any remaining space should be saved for critical features to make sure current 328p users will be able to use them on their machines during the transition to a faster and larger chipset. (FYI, the Grbl-Mega branch is not the final chip, but a stepping stone to make sure that things like this get developed and added, as I continue working on the ARM version.)

But, I'd like to keep this conversation going. Maybe at the grbl-Mega repo. Do you have a formal proposal on what to add and how? Using Grbl's '$' system command character is a good place to start, as it's easy to pick out of a g-code stream and formulate any type of command after it.

J-Dunn, 2016-04-29:
OK, I clicked on the link to the mega port and did not see an issues tab nor a new issues button. Does something need to be turned on there?

chamnit, 2016-04-29:
@J-Dunn : Turned on.

closed 2016-04-29 by chamnit