grbl/grbl #981: Precision for cnc with pololu drivers

Dareblix, 2016-04-29:
Hello everyone, days ago i`ve posted a problem with the steps per mm and they gave me the answer, but i didn`t understood the way he solved the math to do that, but well i`m using 1/4-20" threaded rod   and easy drivers the steps per mm i used it was 1260 (this worked fine) but now i bought a cnc shield with pololu drivers and this steps per mm doesn`t work at all, if you know the answer that would be awesome and help to understand how did you get the answer. Thanks eveyone for read!

cheton, 2016-04-29:
You can try this calculator to fit your need:
http://prusaprinters.org/calculator/

Dareblix, 2016-04-29:
I`ve tried that, but i didn`t understood how to put the threaded rod info

cheton, 2016-04-29:
I just searched from previous discussion, your <b>leadscrew pitch</b> is supposed to be (1/20 * 25.4) = 1.27mm, and you need to know the microstepping value of your Pololu driver, assume you have it microstepping at 1/16.

Just input the following value to the calculator
* Leadscrew pitch: 1.27 (mm/revolution)
* Step angle: 1.8° (360° / 1.8° = 200 steps per revolution)
* Stepping: 1/16

After the calculation, you will get the result:
```
200(step/rev) * 16 / 1.27(mm/rev) = 2519.69 (steps/mm)
```

pth3k, 2016-04-29:
Check the micro stepping jumpers on your old solution and set the pololu
jumpers to the same microsteps.

closed 2016-09-03 by chamnit