grbl/grbl #982: Skipping lines after 1 full buffer of commands

PieterjanDeconinck, 2016-04-29:
Hello all.
I'm currently using the X-Carve controlled by grbl as part of a project I'm working on. It's not used as a CNC machine. Instead, the spindle has been replaced by a pneumatic cylinder. The valve for this is controlled with the M3/M5 spindle control.

So this all works like charm, at least for a while. After a fixed amount of commands, the system starts to behave strange. It looks like it's skipping a lot of lines. I'm using large moves, so this is a big problem for my system. I wrote my own GUI in C#, which uses the simple send > wait for ok/error protocol. The strange thing is all the commands are sent and give an ok message. So it seems like all the commands are received, but something strange happens after.

First I thought the problem was with the use of M-commands, but even with just G-commands the same thing happens. When I made the RX serial buffer on the arduino twice it's normal size (256 instead of 128), the amount of correctly executed commands was doubled aswell. So it seems like something goes wrong when the buffer is filled once, but I've not managed to find a solution to my problem...

This is my code to send a command, which is called repeatedly in a for loop:

```
     private void writeLineToSerial(string text)
        {
            if (_serialPort.IsOpen)
            {
                _serialPort.Write(text + "\r\n");
                string answer = _serialPort.ReadLine();
                if (!answer.Contains("ok"))
                {
                    // Try resending the command (USB data corruption possible)
                    _serialPort.Write(text + "\r\n");
                    answer = _serialPort.ReadLine();
                    if (!answer.Contains("ok"))
                    {
                        // Emergency break!
                        _serialPort.Write("!\n");
                    }
                }
            }
        }
```

Maybe I'm making an obvious mistake, but I can't seem to figure it out... Any help would be appreciated!

chamnit, 2016-04-29:
@PieterjanDeconinck : Grbl uses a ring buffer that is 128 characters. Standard ring buffers have one empty buffer slot to track its state, so there are 127 usable slots in the buffer. So for 256 bytes, it's 255 useable bytes. It's a common problem. FWIW, in v1.0, I'm finally fixing this to mean 128/256 usable bytes.

PieterjanDeconinck, 2016-04-29:
Thank you for your answer. But since I'm using the wait for OK, I would think this would not be a problem. I'm not using character counting, so I don't understand why I have this problem. Grbl waits to send an OK if the buffer is full, right?

It seems like when the buffer is full, the next line is sent but ignored. Is there a way to fix this while still using the wait for ok protocol? Or do I have to change to character counting?

chamnit, 2016-04-29:
@PieterjanDeconinck : If you are sending too many characters, you will overflow the serial RX buffer and corrupt the data, which will cause Grbl to behave much like you describe. Try reducing your character counting limit to 127 bytes, rather than 128.

chamnit, 2016-04-29:
@PieterjanDeconinck : Dang it. Sorry. I really need to stop skimming these messages. I misread your post. Gimme a sec to re-read it.

chamnit, 2016-04-29:
@PieterjanDeconinck : Hmm, not exactly sure what is going on. If Grbl works with other GUIs like UGS or bCNC, then it would be something wrong with your streamer. If this is the case, you can have Grbl echo back the commands it receives to help you diagnose the issue. Uncomment ```#define REPORT_ECHO_LINE_RECEIVED``` in config.h, re-compile, and flash.

109JB, 2016-04-29:
Are you resetting the "answer"  variable after sending? If so, I don't see it.

My VB streamer doe it this way:

For i As Integer = 0 To gcodearray.Length - 1
      GRBLResponse = ""
      SendToGRBL(gcodearray(i) & vbLf)
      Do Until GRBLResponse = "ok"
            Application.DoEvents()
            If GRBLResponse.StartsWith("error") Then Exit Sub 
      Loop
Next        

My variable GRBLResponse would be equivalent to your answer variable

109JB, 2016-04-29:
Sorry, the post editor stripped the indenting.

electrokean, 2016-04-30:
What is your serialport object timeout for ReadLine?
You could try either inserting dwells (G4 P0) between each command, or use check gcode mode ($C) to see if anything is different.
I've written a couple of custom GUIs myself and not seen this problem.  My GUIs are typically event driven state machines, but I've also written code much like your example on a secondary Arduino and it has worked fine.  I do typically have the occasional dwell due to the type of operations I perform.

PieterjanDeconinck, 2016-04-30:
Thx for all your quick answers! I finally found the issue and as expected it was not a big one :p
Apparently, in Grbl there's no difference between LF and CR. Since I was sending "\r\n" instead of "\n", Grbl responded twice with ok. So it's logical the communication went totally wrong, because this way my stream just overflowed the buffer.

@chamnit : Your suggestion was really helpfull, because this meant I had to put a ReadLine twice. This revealed the double ok message. Thx!
@109JB : This is not a problem, since I declared 'answer' as a local variable. This is automatically removed after execution of the function.

closed 2016-04-30 by PieterjanDeconinck

s-fr, 2017-11-06:
Hey!

I got the same issue. Could you please post your working code?

King regards