grbl/grbl #983: Source code, where does the actual digital output take place?

bidaian, 2016-05-02:
I've built a drawbot that uses DC motors from very old CD drives. They are NOT steppers, just simple DC motors with an H-bridge. I modified the very basic interpreter from this project[1] so that it drives the DC motors (setting HI-LO or LO-HI and waiting). And It works!

BUT I'd love to use grdbl and have much better gcode parsing and support. So I plan to do the same, go to wherever the grbl digital output takes place and adapt the code to drive the DC motors instead.

I've checked the source .c files (stepper, planner, etc.), but I cannot find the actual place. Could you please point me in the right direction?

[1] http://www.thingiverse.com/thing:201807

chamnit, 2016-05-02:
@bidaian : It depends, but the steppers are executed in stepper.c. Look at the ISR routines.

closed 2016-05-02 by chamnit

bidaian, 2016-05-02:
Ok, thanks.