grbl/grbl #986: Meaning of "Door : x"

LETARTARE, 2016-05-06:
Hello,
what will be the use of
- "Door:x"   x in [0,1,2,3] ?
- "Hold:x"  x in [0,1] ?
- "Sleep:x"  x in [0,1] ?

Thanks

chamnit, 2016-05-06:
@LETARTARE : The ':x' values indicate sub-states, like holding and ready to resume. There will be an API document written up prior to the v1.0 release to master. While in beta, it's still subject to change, depending on feedback from the community.

closed 2016-05-06 by chamnit

LETARTARE, 2016-05-06:
@chamnit ,
thank you very much.
A little more detail, would provide an opinion.

chamnit, 2016-05-06:
@LETARTARE : Obviously you read the source code. Comments on what they are are there.

splitn2, 2016-05-08:
@chamnit I like this mention of 1.0 ,  bring it on !!! I am not pushing just very interested, keep up the good work mate! My next wee project includes a GRBL small 3 axis cnc to drill skis , its mentioned on my Git as a very preliminary start, I need to get some young brains on the project to do some programming for the project next...... https://github.com/splitn2/DrillSki