grbl/grbl #987: Remember Last Position

xmenxwk, 2016-05-09:
It would be better if it keep save its position in EEPROM. Because as I connect it again, it reset back to 0, 0, 0.

chamnit, 2016-05-09:
@xmenxwk : Thanks for the suggestion, but I strongly disagree and will never do that. Stepper motors are, by nature, open-loop control. You have no guarantee that position is maintained, even if you don't touch your machine between sessions. This is why homing is so important on these types of CNC machines.

closed 2016-05-09 by chamnit

csdexter, 2016-05-09:
This is especially important with microstepping drivers which, by design, "forget" their electrical position on reset and thus the assumption that "the machine did not move" becomes false from the start, even if you did not physically touch it.