grbl/grbl #988: Closed loop with cheap glass scales

DasWookie, 2016-05-09:
I'm using my GRBL controller to mill aluminum on a desktop CNC. When my endmill starts to dull tho, I start to lose steps. Eventually the whole job goes TU, but I was thinking there might be a way around this where if one could detect missed steps, then there could be an alarm threshold where the job could be paused, the endmill could be changed, and then resume the job... currently tho, there seems to be no means of having any closed loop support with GRBL. So, I was wondering what the possiblity might be of having some sort of linear encoder support added. Then the intended position / internal DRO of the machine could be tested against the encoder, which could them alarm out if too many steps / drift started to occur. There may just not be enough clock systems left in the Arduino for this, or this could be my approaching it from the totally wrong end of things (and should instead look into using steppers with integrated encoders) but figured I'd toss the idea out there and see if it's even possible. I've scrapped enough $60 pieces of billet at this point that throwing a few hundred at some cheap Chinese eBay linear encoders like these http://www.ebay.com/itm/261089356161 would be trivial... and I can't be the only person having issues with skipped steps hoping for some sort of possible hardware fix.

chamnit, 2016-05-09:
@DasWookie : Yes, this type of feedback is on the long-term to-do list. However, you're right. The 328p AVR is nearing the end of its capability, and a closed loop system would not fit into it. A short-term solution would be to add it to the Mega version. I don't expect the overhead for encoder feedback to be significant enough to degrade overall performance. I'll bump this feature up in my development notes.

closed 2016-05-09 by chamnit

csdexter, 2016-05-09:
Also, you can implement the servo-like behaviour off-board if need be. Since the task is dead simple: watch two pins encoding "I went one step forward"/"I went one step backward" then compare with what the encoder/glass scale reads and alarm if they disagree, something as puny as an ATTiny 85 could do it without breaking a sweat.

Adding icing on the cake such as "if commanded number of steps differs from read number of steps by less than X (with X being a small number, say at most twice the machine repeatability), then re-issue last step pulse(s) and see if the machine obeys then" is left as an exercise to the reader.

chamnit, 2016-05-09:
@csdexter : That would work too. In the early days of development, I wrote a simple Arduino sketch that did pulse counting to check if the output step position and Grbl's internal position matched. I can't seem to find it anywhere, but it's fairly trivial to write (need to use interrupts). I think if you search for 'pulse counter', there should be some sketches to template from.

DasWookie, 2016-05-09:
So to count the pulses would I just take another micro, like the ATTINY 85, and just solder in a jumper to the driver to count the pulses being sent to the stepper, and share the ground? I'm just trying to be sure I'm understanding the recommendation here on how to solve the issue I'm having. I think once I've gotten a good handle on how to chase this from that standpoint, I can develop the hardware and code up an arduino (maybe just a pro mini) to monitor the system as a separate piece of hardware which could be used until (one day maybe) that Grbl has support for closed loop systems.

I'm totally self taught in EE land (almost all via Arduino) so I'm sure I'm lacking a LOT of general knowledge here. Everything I've learned has been just what I needed for various projects. I have done some arduino based interrupt pulse counting before, but that was using a hall effect sensor for a speedometer.

I figure I may also be able to slave a optical rotary counter to close the loop, like one of these 400p/r encoders: http://www.ebay.com/itm/361402482978 where using either a gear (or timing belt) to the stepper I could then count the pulses to the stepper and the steps actually taken via the encoder... which might work out to be a simpler (and potentially much cheaper) means of tracking actual steps taken. There are plenty of cheap optical rotary encoders which could be used for an opensource solution to help desktop cnc folks (or larger cnc solutions) who might be pushing the boundary of what their desktop CNCs are capable of, or are possibly using their machines for lights out production type jobs like I am... then if a problem were detected it could then trigger a feed hold if steps started to get lost... so at least one has a chance at changing out the endmill prior to scapping their stock, or melting the windings on their spindle (as I've done twice now) due to excess heat generated from a dull or damaged endmill.

chamnit, 2016-05-09:
@DasWookie : Melting windings? I'm assuming that you don't know this, if you do, no offense intended. Just ignore what I'm about to say.

I think you may be running your end mills too fast/hard or using them well beyond their usefulness.

Heat destroys end-mills and you have to be careful about keeping them cool, especially when machining metal. Most 'lubrication' you see with machining isn't actually for lubrication. It's primarily for keeping the tool cool and below the tool's annealing temperature (gets dull and ruins the tool).

You also have to ensure you are running the correct feeds and speeds (feed rate and spindle speed) for the particular tool. These recommended rates are there to indicate the best parameters for the longest tool-life. Any time you are outside of this, you can sometimes drastically reduce tool life.

The machinists at my day job constantly measure and inspect their tools for premature wear and damage (chipping). Otherwise a part that is often several tens of thousands $ in man-hours can be ruined.

DasWookie, 2016-05-09:
Yupyup. Totally agree on all counts. You can never know what somebody's experience is, so assumptions about my being a no0b with machining (after admitting to burning up 2 spindles) are certainly something one could suspect. :) I'm a machinist as well, and have been for 20+ years. I run manual machines tho. Some of these smaller parts I'm manufacturing however are easier with the complex geometries to run on the desktop CNC. Previously I was casting them in aluminum and machining to finish on the Bridgeport, but it was taking a LOT of time. Would that I could have a Haas VMC and run these things with flood coolant, but alas I do not. That's when I started to eye my Shapeoko which I used for wood vCarving, and figured "Hell, folks ARE running aluminum with these things... I'll give it a try!" which is when I started running aluminum on my poor little Shapeoko 2 nearly 24/7. These endmills last a while, but not forever. When the endmill finally goes TU it does so quickly... and if I'm in earshot of the thing I can head over, hit the eStop (of feed hold) and swap the endmill, and resume the job... or if I didn't get there in time I might have to hack up the gcode to start from a safe spot close enough that I don't spend hours upon hours just machining air to catch back up to where it was prior to loosing the endmill.

The parts I'm running that are specifically having issues run for 9+ hours... thus the running in a lights out configuration. I just can't be in earshot of the thing 24/7 and have to run the machine near all day every day to keep up with production... so far, I'm doing it, but it's costing me time with unknown tool life and spindle destruction. I've tried hacking up the gCode to allow regular breaks to inspect the tooling, but issues with homing were causing more problems than I was solving by just running one really long job and hoping the endmill lives. I do inspect the endmills prior to the start, but endmills don't last forever, and I'm still having difficultly figuring out exactly how long they will live. The two spindles I melted died because in the middle of the night, while I was sleeping, the endmill got dull, galled with aluminum, and that was that. In about 20 seconds it goes from machining well to a disaster. If nobody is around tho, it just spins and spins, making more and more heat... eventually it gets hot enough that the windings resin melts, resistance goes to infinity, and POOF there goes the spindle. I've switched over from running the DeWalt DW660 over to the DeWalt DWP611.

I had my speeds and feeds dialed in pretty good with the 660, but I'm still getting the groove of the 611 where I can run the spindle slower so don't need to be AS aggressive with the feeds so my HOPE is that I won't skip steps as often. I just switched over to the 611 yesterday, and went through 4 endmills and still haven't finished the first part. Arghhhh! I might upgrade to larger steppers, but at some point I just need to accept that it's a desktop CNC and it's only going to perform so well. Still, it's what I have. The endmills typically live for about 6-8 parts. Sometimes 4, sometimes 10. It's the same billet, so I don't know why with the same 6061 Aluminum stock, settings, endmills (I buy them 20 at a time) I'm getting such variation in tool life. I don't want to just toss the endmills after 4 parts tho (these endmills are $13 each) so after 4 I start inspecting them closely for signs of wear prior to starting the next part... I've also switched over to starting with a new endmill each evening, regardless of what the one in there looks like, to also reduce the chance of it going TU while I'm in bed sound asleep... but that's no guarantee that the thing won't die in the middle of the first part. {shrug} I'm also swapping over the bed to UHMD sheet so I can run a misting WD40 coolant, also in hopes of increasing tool life. The current MDF bed wouldn't take to WD40 too well tho. ;) I'm running these endmills at 0.020" DOC at about 18 IPM. The spindle speed was 30k which, I agree, was REALLY fast, but that's the only speed the 660 spun at. The 611 is variable speed, and I've been running it at the "3.5" speed (whatever that is) and dialed my feed up to about 22 IPM. The endmill is running at a cool 109F and I'm getting larger chips... but my tool life is WAY shorter than it was running at 30k at 18IPM. Still screwing with it and trying to figure out how hard I can push it with this new spindle. I'm pretty much right on the ragged edge/limit of what these Shapeoko2's can handle tho.

109JB, 2016-05-09:
I would suggest that if you want closed loop operation with GRBL that there are already options available right now.  

1 - Closed loop stepper drivers - There are right now closed loop stepper drivers that close the loop at the driver itself. Not in grbl, or LinuxCNC, or Mach3..., but is the driver controlling that one motor. Here is an example:

http://www.automationtechnologiesinc.com/products-page/nema23-closed-loop-stepper-motor-system-hybrid-servo-kit/hybrid-servo-drive-kl-5080h

2 - Closed loop servo drivers - Same principle as above, where the loop is closed at the driver.

http://www.automationtechnologiesinc.com/products-page/dc-servo-motor-driver/geckodrive-servo-motor-driver-g320x

Both of the above close the loop at the driver with feedback from a rotary encoder detecting the shaft rotation of the motor.

That brings up another point.  Using the glass scales for feedback will not work the way you think, especially in the sense of a servo.  However, it won't work all that well for a stepper even just for missed step detection.

A linear scale is going to be measuring the actual axis motion and not the motion of the motor.  Most  machines do not have a perfect correlation between motor movement and axis movement. Things like sticky ways, slop in the drive system, and inaccuracies in the drive system all come into play. For example, a screw driven axis on a typical milling machine. The motor is connected to the lead screw by a coupling that can have play. The lead screw and nut can have play (Backlash). Unless the lead screw is a very expensive precision ground screw there will be lead errors. What does all this mean?  If you command a 1.000" movement, the stepper may actually move the proper amount based on the steps/inch, but the axis only moves 0.998 inches. There are no missed steps but the linear scal will think there are.  The better, and easier option is to use a rotary encoder to do what you are talking about.

Another point about the linear scales. Even in production machine that use glass scale feedback, they first use a rotary encoder at the motor in a dual-loop feedback system. If you have any slop in the system and were to try to close the loop using only the linear encoder the motor would dither back and forth constantly due to the slop in the system. 

Bottom line is if you want a missed step detection system, use a rotary encoder mounted directly to the stepper. If you need a rotary encoder that is relatively cheap and is proven to work then here you go. This can be mounted directly to the back of a dual shaft stepper motor.

http://www.automationtechnologiesinc.com/products-page/servo-motor-encoders/cui-amt102-v-kit-encoder

rwensley, 2016-05-09:
@DasWookie I have been using the 611 for several months now with great success, after figuring it out that is.  I gutted the internal speed controller and added the SuperPID controller so that I could have better control of the speed. The speed of the 611, as well as any universal motor, is highly non-linear.  A change from "1 to 2" might be 1000 RPM but from "2 to 3" it might be 1700 RPM.  With the external speed controller you can compensate for it.
If have run a fair amount of cast aluminum 6061 through my machine so far.  Nothing that runs for 9+ hours at a time however.  For 6061 I run the spindle at 17,500 rpm with a feed of 41.8 IPM and a plunge of 21 IPM the DOC is 0.0118" for roughing.  For finish the settings are the same except the DOC is 0.0023". The end mill is 2 flute 0.125" with 1" stickout.  So far they are lasting a long time.  I do not know the exact time however.
If you need more info you reach me at ronwensley@live.com
Cheers,

langwadt, 2016-05-09:
@DasWookie how about putting a current sense or temperature limit on the spindle instead since that is the real issue? I've never had much luck even drilling alumnium dry, but just a drop of ethanol prevent the chips from "welding" to the tool and it cuts like butter

rwensley, 2016-05-09:
@langwadt  Ethanol as a lubricant?  Please tell me more....The only ethanol I am familiar with has a very low vapor pressure (ie gasoline).  I am interested in learning about this.

langwadt, 2016-05-09:
@rwensley ethanol, the alcohol you can drink ;) though you for this you'd use the cheap denatured alcohol you can't drink. You need very little of it and it evaporates without smelling or leaving a mess like wd40, and imo it works better

rwensley, 2016-05-09:
Oh yea Ha... It's ethylene in gasoline...my bad. I had never considered using denatured alcohol.  I would have thought the low vapor pressure there as well would not make a good lubricant. Interesting, I'll have to give that a spin.

AlexHolden, 2016-05-10:
Misting WD40 and ethanol both sound like a bit of a fire hazard for unattended operation. Not tried it myself, but I've heard good things about a product called Kool Mist.