grbl/grbl #989: Work with One stepper motor but not 2

Kingdombed, 2016-05-09:
Hello everyone !

I have a strange problem here : I'm using a CNC shield arduino with steppers motors and steppers drivers. The problem is everything works fine when I only plug one motor but , as soon as i plug the second motor, both don't work anymore. They are making high-pitched sound but they don't turn anymore.

Here some photos of my project and what kind of component I use : 
[image]
[image]

I'm using : 
-An external power source,
-An Arduino CNC shield with StepStick DRV8825 as stepper drivers
-An Arduino UNO
-2 42BYGHW609 as steppers motors

What I tried for the moment :
- Changing the GRBL version : I tried 0.9c and 0.8 but no change
- Bring more power : No change, but I only put 16V max
- Testing the stepper drivers one by one : They both work well when I use them separately.
- Changing the config of GRBL to make them turn slower : Still no change.

I'm running out of idea so I will be glad if you have some ^^
Thank you in advance for your help

electrokean, 2016-05-09:
First, try heavier wires direct from the power supply to the driver board.
Also, watch how much current is being drawn from the supply - maybe it is going over the max current limit.

jahnj0584, 2016-05-09:
whats the PSU?

electrokean, 2016-05-09:
https://en.wikipedia.org/wiki/Power_supply_unit_%28computer%29

electrokean, 2016-05-09:
You can see it in the photo - seems to be a 3A model.
3A may be too little if he is running two motors at 1.7A each - especially with such thin wire, with breadboard connections in between

jahnj0584, 2016-05-09:
oh, I know what it is. I mean, what model is OP using? I have 24V for my 4 motor 3axis OX build. I've seen 12V be used if you only use nema 17s.

electrokean, 2016-05-09:
@jahnj0584 apologies for initially misunderstanding your question :(
The OP seems to have his power supply set to 12V, and it seems to be a 3A unit.  With those thin wires and breadboard connections, he will get even less voltage to the drivers once they draw any significant current.
Not sure if the high-pitched sound OP describes is just the holding current PWM frequency, or the motors actually stalling.  Probably the latter though.  If heavier wires, beefier PSU, or [pssibly a higher voltage don't help, then he may need to look at reducing his acceleration setting.  But that would typically be noticable with one motor, so I really think OP's power supply and wiring isn't up to the task.
A higher voltage could help provide a higher acceleration ramp without stalling, depending on motor inductance.

109JB, 2016-05-10:
Have you adjusted the current output of the DRV8825's?

Kingdombed, 2016-05-10:
Hi again !
Thanks for all your replies, I'm french and i'm trying to understand everything. 
I will try to use better cable next time to see if it change anything.
I've tried to max the input power of my power supply but It don't work. I will try to find a better power supply.

> Have you adjusted the current output of the DRV8825's?

I don't know how to do that. I assumed that the CNC shield would do it for me.

X3msnake, 2016-05-10:
https://www.pololu.com/blog/484/video-setting-the-current-limit-on-pololu-stepper-motor-driver-carriers

If your drivers are clones beware do not use the via of the driver as
mesure point only the screw as it is usually a diferentes layout from the
original ones

Kingdombed, 2016-05-23:
OK it work.I just need to put more amps but the arduino uno died. Overheat. Thank you

closed 2016-05-23 by Kingdombed

X3msnake, 2016-05-24:
It shouldn't die

- You should not connect or disconnect the motors with the arduino on
- You should use separate power sorces for the arduino and for the drivers
- Only the ground of both power supplies must be shared otherwise drivers
will not work

jaketeater, 2016-05-24:
I am not sure if this was pointed out, but the steppers are not connected to the shield in the same order in the picture above.  The green and black wires are inverted.  Its possible that the coloring's are different and the phases are wired correctly, but it may be a good idea to recheck the wiring.

Get a multi-meter and test for continuity.  Make sure the wires with continuity are placed on adjacent pins on the shield.  This video gives a good explanation of finding the phases:  [https://youtu.be/tAusnFPgOcA?t=1m2s](https://youtu.be/tAusnFPgOcA?t=1m2s)