grbl/grbl #990: Cooling cycle until task done

julienja, 2016-05-11:
I am new on grbl,
I do some engraving, I have so project for large engraving, it could take up to 4 hours.
But I don't want to burn my laser.

Can we make a command respecting the cooling cycle
It will look like:
Work 40mn/cool 30mn laser inactive/restart for 40mn. Until work completion 100%
  No need for home position, so I don't lose track, and no shift in engraving.

How the command will look like in the left panel?
P30000 M03 M01?

Thank you for your help

chamnit, 2016-05-11:
@julienja : There isn't a need to add a new g-code command. Everything you need is already available for you. The G4 Px command will pause the program for x seconds. Or just cool your laser better so you don't have to pause.

closed 2016-05-11 by chamnit

julienja, 2016-05-11:
OK, but do I put it the commands before I launch the engraving? Because during  the process I cannot other commands?

Thank to be always so quick

julienja, 2016-05-11:
I put my laser under a big fan controlling temperature, the laser died under 18°C.
Is there a way to create an automatic pause, with a timer, so the machine can work on its own?

tklus, 2016-05-11:
Are you driving your laser at too high of current? I have ran my laser for as long as 9 hours with no pause. I am running a 3w laser on my machine with a printed shroud and small fan. Make sure you are not exceeding the rated amps of the laser.
[image1.JPG]

julienja, 2016-05-11:
I use a 500mw laser, I bought a kit online, how can I verify it?

Thanks

tklus, 2016-05-11:
Can you send me the link? Mine is a 3 watt running at 6.5 volts and 1.6 amps.

This is the laser I am running. Towards the bottom of this page describes tuning the driver
http://3dpburner.blogspot.com/p/lectronic.html?m=1

julienja, 2016-05-11:
http://www.dx.com/p/neje-500mw-desktop-violet-laser-engraving-machine-printer-diy-kit-silver-transparent-380428
 
But I am  covered by the guarantee, the cooling cycle important more, and  make the laser faster or Les powerful it goes through leather

tklus, 2016-05-11:
I would suggest you check the voltage output to the laser and then check the current. Depending on what the voltage is at, you can set the current to not exceed the rated 500mw on the laser. For example, if the laser is running at 12v you would set the current to 40milliamps. Wattage divided by voltage equals amperage. So take .5 watts divided by 12v equals 40 milliamperes or .04 amps.

Hopeful that makes sense. If not, just try turning the power down bit...

I

http://www.dx.com/p/neje-500mw-desktop-violet-laser-engraving-machine-printer-diy-kit-silver-transparent-380428

tklus, 2016-05-11:
That 40 milliamperes number might not be right. Mine runs at 1.6 amps and 6 volts for a 3 watt laser.

Not sure what yours should be. Maybe contact the vendor and see what they recommend.

julienja, 2016-05-12:
I contacted the manufacturer, with no response.

Is there no command I can program  to cycle cooling? I tried to find more info on the command G4px 
Should I edit my NC file and input the G4px command? Can I program the pause  before to launch the engraving? So I don't have to stay around the machine

csdexter, 2016-05-12:
@julienja you appear not to have read/not to pay attention to what previous posters explained to you: there is no support within the G-Code language for implementing what you want and it would be the wrong place to implement it in in the first place.

The G-Code language supports a "flush motion queue and wait X seconds" command but not a pre-programmed repeated pause. You may be able to instruct your CAM post-processor to insert such pause commands in the G-Code stream, but it's going to be neither easy, nor good looking.

The right way to approach this problem and fix it is to fix your laser, by either:

1. making sure it is current ran well within its safe operating area
2. reducing the load so that it runs closer to the bottom of the safe operating area
3. implementing better or more efficient cooling so that thermal excursion does not occur during a machining cycle.

julienja, 2016-05-12:
OK @csdexter 
I didn't misread, but it wasn't obvious and clear for myself. I put my laser under a big fan checking temperature not exceeding 20°C.

The manufacturer noticed  the laser can work 40mn max with a cooling of 30mn. I wish to run it in cool environment, but I will be out of warranty. That's why I came asking a cooling cycle on grbl.

Maybe the flush motion queue is my only solution ?

Thank you

csdexter, 2016-05-16:
If grbl supports a feed hold input (either via command or via a button connected to a GPIO), you could use that with a timer. You start the timer the first time you turn the laser on and when it hits 40 minutes, you issue the feed hold command. 
If a feed hold alone doesn't also stop the spindle (which is the laser in your case), you also inject a spindle stop command right after Grbl reports that all movement has stopped (as commanded by the feed hold). Alternatively (and possibly cleaner, since you don't interfere with the command stream), just turn the laser off externally (at the timer, for example).

The timer then starts counting the 30 minutes for the mandatory pause. When that expires, it reloads with the 40 minute value again, releases the feed hold, tells Grbl to resume (cycle start?) and re-enables the laser.

julienja, 2016-05-16:
Excellent, how do you write it down 
~ g4p x(x for seconds)

julienja, 2016-05-25:
Csdexter
Do you know how the code will looked like?  Thanks

electrokean, 2016-05-25:
Just insert an extra line like this wherever needed in your code
G4 P300
this would pause for 5 minutes (300 seconds) before continuing
You'll probably need to include M commands before and after that turn off and on the laser (spindle)
Note that the maximum pause value is 3276.75 (about 54.6 minutes)

langwadt, 2016-05-25:
back to start, those time frames makes no sense, unless there is a huge mass involved there is no way it takes 40 minutes to reach too high a temperature or 30 minutes to cool down

julienja, 2016-05-25:
I follow the constructor instruction. I have already one laser burnt