grbl/grbl #992: $C (check toggle gcode mode)  s/b (toggle gcode mode)

Chips-cnc, 2016-05-16:
The output of the “$” command for “$C” should be changed from 'check' to 'toggle':

Grbl 0.9j ['$' for help]
$$ (view Grbl settings)
$# (view # parameters)
$G (view parser state)
$I (view build info)
$N (view startup blocks)
$x=value (save Grbl setting)
$Nx=line (save startup block)

$C (toggle gcode mode)

$X (kill alarm lock)
$H (run homing cycle)
~ (cycle start)
! (feed hold)
? (current status)
ctrl-x (reset Grbl)

Similarly, the documentation on web page:
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9
should be changed (and the word 'and' added between blocks and process):

$C - Toggle gcode mode
This toggles the Grbl's gcode parser to take all incoming blocks *** and *** process them completely, as it would in normal operation, but it does not move any of the axes, ignores dwells, and powers off the spindle and coolant. This is intended as a way to provide the user a way to check how their new G-code program fares with Grbl's parser and monitor for any errors (and checks for soft limit violations, if enabled).
When toggled off, Grbl will perform an automatic soft-reset (^X). This is for two purposes. It simplifies the code management a bit. But, it also prevents users from starting a job when their G-code modes are not what they think they are. A system reset always gives the user a fresh, consistent start.

chamnit, 2016-05-16:
@Chips-cnc : Feel free to update the Wiki page for typos, as it is a wiki after all. It's faster and easier for everyone when people take the time to correct these types of things, rather than post a long note, wait for a response, and see if someone else will do it.

closed 2016-05-16 by chamnit

Chips-cnc, 2016-05-16:
chamnit - I have added the word 'and' to the doc (as my first wiki change ever) but I have NOT changed the 'check' to 'toggle' because (in my opinion) the code that generates the output of the of the '$' command needs to be changed first (otherwise, the doc will be out of sync from the code).

Since I am new to this, what should I have said so that someone would realize that the code needs to be changed?

chamnit, 2016-05-16:
@Chips-cnc : Thanks for updating the Wiki. As far as the 'toggle' vs 'check', the source code is the way it is and I'd rather not change it unless I have to. There is a chance that some GUIs may be broken by altering Grbl's feedback in any way. It's better to just update the Wiki. Besides, it's just semantics. 'Toggle gcode mode' vs. 'check gcode mode' is no more clear than the other. Both are somewhat ambiguous with the latter being arguably less so.