grbl/grbl #993: Wondering if G40 G41 and G42 are on the horizon?

rfresh737, 2016-05-17:
I was just wondering if G40, G41 and G42 are on the developers to-do list at some point down the road? Radius Compensation.

I saw a post about it here (its closed now) but it was never made clear if these are or are not on the to-do list?

Thanks...

csdexter, 2016-05-17:
@chamnit can confirm or correct me, but it's most probably one of the math-heavy features that did not fit either in the remaining flash space of the '328p or in the remaining timeslots while Grbl is running.

cri-s, 2016-05-17:
no, thats is not the problem, at least for older 0.9g firmware, offset
compensation and canned cycle fit inside the 328p.

chamnit, 2016-05-17:
@csdexter @cri-s : You're both right. Canned cycles/tool radius comp do take a bit of room, but I've elected not to. They are not critical features and are set for later release. (However tool compensation is more difficult to fit because you have to have tool information stored on-board as well.) However, any remaining space is reserved for new critical features, such as overrides, jogging mode, and general clean-up for the v1.0 release.

closed 2016-05-17 by chamnit