grbl/grbl #994: Change Baudrate

othtekki, 2016-05-17:
Is it possible to change the baudrate of GRBL v0.9i to =<57600?
My problem is the serial wire is longer than 2m and the baudrate 115200 is for wires <2m.
I know downgrading to v0.8 is possible but i would rather keep v0.9i.

Thanks for read.

cri-s, 2016-05-17:
Instead of
  // Set baud rate
  #if BAUD_RATE < 57600
    uint16_t UBRR0_value = ((F_CPU / (8L * BAUD_RATE)) - 1)/2 ;
    UCSR0A &= ~(1 << U2X0); // baud doubler off  - Only needed on Uno XXX
  #else
    uint16_t UBRR0_value = ((F_CPU / (4L * BAUD_RATE)) - 1)/2;
    UCSR0A |= (1 << U2X0);  // baud doubler on for high baud rates, i.e. 115200
  #endif
  UBRR0H = UBRR0_value >> 8;
  UBRR0L = UBRR0_value;

  // enable rx and tx

i use:

#define BAUD BAUD_RATE
   #include <util/setbaud.h>
UBRRH = UBRRH_VALUE;
UBRRL = UBRRL_VALUE;
   #if USE_2X
   UCSRA |= (1 << U2X);
   #else
   UCSRA &= ~(1 << U2X);
   #endif

109JB, 2016-05-17:
To answer the original question.... There is a setting in the config.h file for baud rate. You just change the value before uploading grbl to the Arduino. I don't have it in front of me right now, but the line is something like

#define baudrate 115200

koppi, 2016-05-17:
https://github.com/grbl/grbl/blob/966abff3cc7be25f2891a8b234bc91d8eddc36a8/grbl/config.h#L36-L37

closed 2016-05-29 by othtekki