grbl/grbl #995: Atmel FLIP w. 16U2 Problem

OldSalt1945, 2016-05-19:
When I try to "Load HEX File" with FLIP, I get "Address is out of Range" error.

I downloaded this HEX file:   https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/firmwares/atmegaxxu2/Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3.hex

I'm stuck.  I don't know enough to go any further. I've spent hours searching the net to no avail.

I'm trying to cure the USB to Serial transmission errors problem outlined here:   https://github.com/grbl/grbl/wiki/Known-Bugs

I am using Win 8.1 communicating with a genuine Arduino UNO R3 recently purchased from Radio Shack.  It has the 16U2 chip.

Please help.

John

OldSalt1945, 2016-05-19:
Forgot to mention, I am running GRBL V 0.9j.

109JB, 2016-05-20:
You have to reset the 16U2 as shown in the link below. 

https://www.arduino.cc/en/Hacking/DFUProgramming8U2

wega52, 2016-05-25:
Hi, i had the same problem,, solved it by installing the Usb Driver which comes with the FLIP Software

OldSalt1945, 2016-05-26:
The problem is that I don't know how to install that driver.  
I think it is the "AtLibUsbDfu.dll" file, but I don't have any idea of how to go about installing it.  
When I run FLIP, it complains that "AtLibUsbDfu.dll" is not found.
Would you mind giving me some detailed instructions on how to go about installing this driver?

John

wega52, 2016-05-26:
Open Device Manager
Look for "unknown device"
if there is no unknown device plug in the UNO Board and  reset the 16U2 not the UNO
the device should show up
right click the  entry and select "update driver"
select  "search for driver on computer"
navigate to "C:\Program Files (x86)\Atmel\Flip 3.4.7\usb"  or where you installed FLIp
confirm the directory
the computer  will find the correct driver and install it
Done
The problem with the dLL will be  solved too

Werner

OldSalt1945, 2016-05-27:
wega52,  thank you ever so much for the explicit directions.  I was able to get Flip running and USB connected.

I am now ascertaining that I have the proper driver to flash to the 16U2.  I am pretty certain that I will use Arduino-usbserial-UNOR3 by Alex Holden.  Just a couple more things to check before I commit.

Finally, I am off the rock and have my wheels on the ground.  Might even make some progress.

Thanks again for taking the time with me.

John

OldSalt1945, 2016-05-27:
OK.  I determined Arduino-usbserial-UNOR3 is the one I'm after.

Now I am confronted with Flip which presents a variety of options, none of which is familiar.

I did Load Hex File.  But now I am unsure.

Do I Erase, Blank Check, Program, Verify, Run, Select EEPROM, Start Application.
So many opportunities to see the magic smoke released.

Help again, please.

John

OldSalt1945, 2016-05-27:
Flip had Erase, Program, and Verify checked, so I clicked Run.  It appeared to work.

If I did wrong, let me know.

John

dieterg2, 2016-07-30:
John,

at the moment I'm facing the same problems as you in may. Have you used an ISP programmer to get the hex file loaded? Or is it sufficient to connect the Uno R3 via it's "normal" usb connector?
The ISP programmer I'm using, and I just bought it for the programming of that 16U2 problem, is installed well in the device manager (as usb device, no COM interface), but Flip can't open the usb connection!?
Any help is appreciated.

Dieter

OldSalt1945, 2016-07-30:
You don't need the ISP programmer to load the hex file.  Loading the hex file is done using the normal USB connector.

When following the instructions at https://www.arduino.cc/en/Hacking/DFUProgramming8U2
you have to reset the 16U2 as is described under the heading "Reset the 8U2".  I was confused by the title and didn't do this as I was using a 16U2. 
If you don't do this step, nothing works.
The title should read "Reset the 8U2 or 16U2".
Once you do this, you will see the 16U2 on the device manager and it will be available on Flip.

This drove me nuts until 109JB told me to do it.  Simple, silly little crack in the sidewalk that trips you up.

John

dieterg2, 2016-07-31:
Thanks, I was able to load the hex file now. It was confusing for me that I read in a german forum that 16U2 could be programmed using an ISP programmer only. But after reading the arduino article again I found the statement about the usb connection.
But my arduino still reports errors like invalid gcode ID:25 or bad number format, even if the gcode check tells me, that everything is ok.
Still some hours to spend for investigation.

109JB, 2016-08-01:
What program are you using to send data to the arduino?

dieterg2, 2016-08-02:
I have used a simple test interface from a friend. The good news is, that he found the problem. Now everything is running well.

ygorma, 2020-07-11:
> Open Device Manager
> Look for "unknown device"
> if there is no unknown device plug in the UNO Board and reset the 16U2 not the UNO
> the device should show up
> right click the entry and select "update [...]

Definitive solution.

M-83, 2021-08-15:
> Open Device Manager
> Look for "unknown device"
> if there is no unknown device plug in the UNO Board and reset the 16U2 not the UNO
> the device should show up
> right click the entry and select "update [...]

I've spent 10 hours over 2 days on dozens of websites to find this comment.  It actually fixes "cannot open usb device".  
Thanks!!!!