grbl/grbl #996: How do a change GRBL for a corexy system

Art999, 2016-05-20:
How do a change GRBL for a corexy system.  When I edit the config.c or .h file the new master is rejected by the Arduino IDE for instllation in the library.
Art999

Protoneer, 2016-05-24:
Sorry for asking the obvious... Have you followed the steps in [https://github.com/grbl/grbl/wiki/Compiling-Grbl](https://github.com/grbl/grbl/wiki/Compiling-Grbl)

If you did it might just be that you are editing the wrong files. (Most likely the library used by Arduino is located in a funny path.)

Art999, 2016-05-26:
Yes, I followed those instructions.  It didn't work.  

I think I finally got it to work BUT not by following those instructions AND I am not confident that I did ALL the right things so that all the GRBL features as built in would work for corexy.
Art99

Art999, 2016-05-28:
I have ONE Arduino that I somehow managed to program that DOES work in COREXY.  I can't figure out what I had done to succeed.  When I try to do what I thought was correct IT DOESN'T work.  GRBL is loaded and responds appropriately BUT when I give it an X10 command it travels on the diagonal, which indicates it is NOT doing the COREXY calculation.
What I have done is as follows"
1)Download the master Zip file
2)Extract all components
3) Modify the config.h file using Wordpad.  Specifically, I UNCOMMENT the line that reads #define corexy.
4) Copy the grbl file from the extracted master with the modified config.h file into the Sketchbook Library.
5) Open the grbl Example "UpLoad"
6) Uploaded this file to the Arduino.

Grbl works correctly on commands like #, $$, $X100=20, but on x20 it moves the gantry on the diagonal.

WHAT am I doing wrong and what should I do (and have probably done once by accident) to get the   system to work in corexy.

Art

Art999, 2016-05-29:
TIME FOR ME TO EAT CROW.  The procedure that I described in my prior comment does work!  I now believe my difficulties prior were due to a defective Arduino.
Art

closed 2016-05-29 by Art999

MatrizCAD, 2018-01-10:
What is the version of your GRBL?

imanelec, 2018-03-29:
I am also having the same problem. if you find the setting for config.h please let me know.

Ianmcmill, 2018-12-13:
Be sure to edit the correct config.h from the added library. Don't edit the downloaded config.h but the added config.h in you Arduino IDE.
Find the correct config.h in your sketchfolder/libraries/grbl.
You can find this directory when looking in the Arduino settings "Sketchbook directory/location".