grbl/grbl #998: GRBL Windows simulation.

usbcnc, 2016-05-23:
This is my attempt to learn how grbl works and I compiled with Visual Studio 10 (quite a bit of porting since Visual Studio does not like some style in c). The exe file is only 40KB. My goal is to see if I can port to other platform like STM32F103C8T6 board which is sold for about $2 on Aliexpress. That module has USB and running at 72MHz. It also has 64KB flash. If it works it should provide better performance.

The simulation helps me to debug the code.

To execute the exe alone (console mode), type grbl and you issue command commands directly on console. This is warm up exercise and real fun starts with a true serial port parameter. 
[image]

If you execute command with com port (limited to 10) you can use Universal-G-Code-Sender-master to send. (eg. grbl COM7, make sure the comport is available or it will go back to console mode.

In the two computer setup, you can use one computer send gcode and another computer to run grbl.exe and use null-modem cable to connect them.
In the single computer setup (my case), I use a USB hub and TWO USB to serial module connected with null modem cable. Ideally this should work exactly like on Arduino target. I have tried the comport simulation free version and it has some issues.

If this is not the proper place to discuss this topic, please delete this issue.

[grbl.zip](https://github.com/grbl/grbl/files/277975/grbl.zip)

chamnit, 2016-05-23:
@usbcnc : Thanks for sharing. If you didn't know already, there is a grbl-simulator that will compile grbl's source code to command line, much as like what you have done. Also, please note that Grbl's GPL license requires you to share the source code.

closed 2016-05-23 by chamnit

usbcnc, 2016-05-23:
Source code attached. New to github. The STM32 part is not working yet.  Use thread for timer0 and timer 1. Also use thread to handle serial port receive and send or console input.

[grbl-masterwin.zip](https://github.com/grbl/grbl/files/278036/grbl-masterwin.zip)

ashelly, 2016-05-23:
It will be interesting to compare this to the existing sim.