grbl/grbl #1000: backlash compensation on GRBL? Please help

bjao1, 2016-05-25:
Hi Everyone,
wondering if anyone has added backlash compensation to their GRBL code. I have a milling machine and my cnc controller used the old pc w/ parallel port mack3 software which has backlash compensation. I got bCNC RPi3 and GRBL arduino working to hopefully replace the old pc setup. My mill has 19mil on X & 15mil on Y axis backlash. Without compensation the circles are not right.
Please help!
Thanks

Church1182, 2016-06-19:
I have not gotten there yet, but I am looking into the same thing. You are using the same setup as I am with bCNC, GRBL, and a RPi3. I'm documenting my build thus far on [YouTube](https://www.youtube.com/playlist?list=PLE5zitZ-arpXyMiXBhZvJYoJkLfbfJNoE) and if I figure it out I'll post it there as well. Part 5 was the first run milling and I discovered that there seems to be 0.010" on the x-axis and 0.005 on the y-axis.
When I get it figured out, I will make a video showing how to measure the backlash and then how to compensate for it.

closed 2016-09-03 by chamnit

ruudvetter, 2018-07-29:
Hello Church1182 ,
I am very currious for your video to measure and compensate the backslash in GRBL.