grbl/grbl #1002: limit switches

auditofi, 2016-05-27:
hiii
could any one tell me how to connect  limit switches to grbl shiled v3,and how to configuration with grbl.. for example how to connect limit switches 3 pin GND,OUT,&+5V. to X+&X-..how to connect with the board ??????please Explanation with the diagram..
Regards

109JB, 2016-05-27:
Try the manufacturer's website first

http://blog.protoneer.co.nz/arduino-cnc-shield/

auditofi, 2016-05-27:
I read the all the manufacturer's website and I did not find any instructions about what i need.
thanks @109JB

OldSalt1945, 2016-05-28:
At what stage of installation are you?  Do you have switches in place and the wires ran to the shield, but need instruction on how/where to attach the wires to the shield?  Or are you asking how to install the switches on your machine?  Or do you want limit switches, but don't know where to start?

John

auditofi, 2016-05-31:
@OldSalt1945 I need how/where to attached the wires to the shield both x+- &y+-

OldSalt1945, 2016-05-31:
I am familiar with microswitch type of limit switches.  It sounds like you may have proximity switches or something similar.  Sorry, you will have to do some research for that.

Microswitch switches have COM, NO, and NC terminals.

closed 2016-09-03 by chamnit

Encarnacao, 2016-09-04:
You just conect the switches to X- or X+ to NO and the gnd to COM and should be it
setting on X+ or X- it should be same if u put a multimeter to see continuitys you will see they are the same