grbl/grbl #1005: GRBL support for Sparkfun's AutoDriver

AgBmed, 2016-05-31:
Hi,

I'm currently starting to put things together for a CNC using NEMA 23 steppers at 24V and 3000mAh per stepper... and I was looking into using existing AutoDriver drivers that I have... Not that I got anything against the GRBLShield, but I already had these as well as an Arduino MEGA 2560... I've been reading a bit the forums here and issues and I can see the Arduino MEGA works (mostly), so I'm just wondering if anyone's had experience with those drivers?

Thank you kindly!

langwadt, 2016-05-31:
I'd say that it is impossible to use the L6470 with GRBL, it uses a special voltage mode drive that is automatically adjusted with speed when using the internal motion engine, but when driving it in step clock mode, which you have to for CNC, the chip thinks the speed is always zero.

If you want to use an ST driver you need something like L6474
Theses boards are 1/3 of the price of the autodriver and can be made to work with GRBL
http://www.st.com/content/st_com/en/products/ecosystems/stm32-open-development-environment/stm32-nucleo-expansion-boards/stm32-ode-move-actuate-hw/x-nucleo-ihm01a1.html

chamnit, 2016-05-31:
Just a point of clarification. GrblShield is not affiliated with the Grbl project, whatsoever. It's a separate product, manufactured and sold by a private company. Only products that have a Grbl logo on the boards are officially supported by Grbl.

AgBmed, 2016-05-31:
Well... while it's disappointing I do appreciate you answer! I looked into the ST board you mentioned, it looks quite good, I'm going to see who sells it locally first... so it can drive up to 3 stepper motors at 3 A... 
Do you have any idea how to make it work with the GRBL? I'll simply flash grbl into the Arduino and then everything else just works? lol... 

I wish it was as simple as plug and play... but I'm sure as most things in life, good things take time. 

Thanks again for your quick response!

gerritv, 2016-05-31:
The first hint about the L6470 would be: Unlike most stepper motor drivers, the dSPIN is controlled over an SPI link. So no for Grbl.

You would need 3 of the boards that langwadt mentioned to drive 3 motors. Personally I would think there are easier to implement solutions for the same or less money. Locally Creatron has some selection of driver boards but not at 3A.
I use a printer-port/Mach3 type board with 4 axis, uses TB6560 chips. There are a variety of those available on eBay, just stay away from the blue boards. The single channel ones provide the best flexibility.
Regardless of the board/driver you will need to spend time tuning the motor step rates and acceleration settings for each axis.

AgBmed, 2016-05-31:
Yeah I visited the getting started PDF... it also requires the main controller board aside from the 3 extra boards.... so I would say there IS a better way... I guess my steppers will have to be underpowered using the grbl shield, but at least I know they work... I believe it's 2.5 amps? so that's prob the easiest way to go... 

I also do have one of those TB6560 boards which I got with my NEMA 23 steppers... but I do not have a parallel port laying around... I moved from the US to Canada and started a new, so I don't have anything old like an old Dell with parallel port... to make things worse I moved from windows to Mac... since windows 10 just sucks... (yeah I know that's a topic for a different forum).... so yep no parallel port... Also, MACH 3 requires windows so none of that for me... 

Thanks for the response, I shall select this as answered! Thank you all for your quick responses!

closed 2016-05-31 by AgBmed

gerritv, 2016-05-31:
Welcome to Canada!
I drive the TB6560 from my Arduino Uno running Grbl. Works just fine. It is just some wiring. I created [GrblPanel ](https://github.com/gerritv/Grbl-Panel/wiki) to drive it.

As for Windows 10, it is much better than most ppl will admit. Too much unfounded nonsense being spouted. I use it for everything on 4 machines, esp Visual Studio development for Grbl, MySensors (both Arduino), and GrblPanel (VB) as well as a few other non-public projects. Computers are a tool, you need to use what supports your needs.

langwadt, 2016-05-31:
@AgBmed if you have TB6560 board that is easy , just replace the parallel port with an Arduino 
all you need is a connector and some wire