grbl/grbl #1007: GRBL laser mode

tklus, 2016-06-06:
Just curious when laser mode will be available for release? I am currently running jtech grbl 9g on my raspberry pi2 with bcnc and it seems that version of firmware doesn't work very good with bcnc. I am using the machine for laser engraving.

Thanks,
Tim

ianclemow, 2016-06-21:
I am using release 0.9j with good results, having set up and compiled GRBL to use the S command for laser power control. unfortunately the best software I have found for producing PWM greyscale pictures  (Inkscape with the 305 Engineering - Raster 2 Laser GCode Generator extension) has a preset maximum greyscale resolution of 256 whilst the default spindle speed max setting in GRBL 0.9j is 1000. Thus the GRBL config.h file  needs to be edited to change the default to 256. It would be far better if the spindle speed settings could be moved the Defaults.h page and controlled with a $ command thus allowing changes to be made without the need to recompile each time to suit the software in use.

chamnit, 2016-06-21:
@ianclemow : In v1.0 alpha in the edge branch, there is already spindle rpm max and min settings available. 

@tklus : All I can say is soon and please be patient. I know its been a while, but Grbl is something I work on in my free time, which I have had very little of recently.

closed 2016-06-21 by chamnit

tasiv, 2016-06-21:
So, if we don't use spindle speed control for standard CNC functions (dewalt router installed) do you think there would be a problem using the current release (0.9j) and modifying the config.h file?  I'm assuming that this method would allow for both CNC and laser burning without firmware changes, right?

ianclemow, 2016-06-21:
Happy to wait

chamnit, 2016-06-21:
@tasiv : Yes. You'll be able to alter the spindle speed change behavior to either stop or continue. Which, in effect, is laser mode off or on.

tklus, 2016-06-21:
Thank you Sonny! I am also happy to wait. I hear ya on the free time thing :) I truly appreciate all the time and effort that has gone into GRBL thus far.

Tim

ianclemow, 2016-06-21:
I can use any PWM port on a standard Arduino Mega card but on Nano and Uno cards PWM function is only available on digital pins 3,5,6,9,10,11. D9 or D10 could be used as a PWM laser control pins but with the loss of Y or X axis limit. (not a problem to many laser engravers).  A better solution would be to use A6 & A7 as the X and Y  limit switch inputs thus freeing up D9 & 10 for use as PWM controls but this would involve more wiring changes.
I am currently modifying the config.h file and recompiling to adjust the laser power profile to suit the particular grey scale characteristics and burning properties for the job in hand. Not an ideal solution as I am sure these features could be updated at job run time (I may be wrong here and there may be a lot more involved than it appears).
 
Ian Clemow

DrZ1968, 2016-06-21:
I'm using Ramps 1.3 w/marlinkimbra fw and 10000 shades of gray = 8bit
engraving with turnkey tyranny plugin for inscape and prontorface. Also
3dburner works well too

DZ