grbl/grbl #1008: GRBL communication

Krenail, 2016-06-06:
Hi, during the past months i built a CNC router with an Arduino Nano and GRBL.
I decided to make a jogger to easily move the router without having to get mad with CAD drawings for simple things.
The jogger use another Arduino which communicates with GRBL trough the Serial port. (RX to TX with a 1Kohms resistor, and vice-versa).
The only problem i got is that the communication between the two Arduinos doesn't seem to work, i tried to run a simple sketch that just send to GRBL a "?" command and get the response printed on a LCD, but nothing appears on the LCD. The strange thing is that it works if i send the message with the serial monitor on the pc.
I've never really understood some things in the Serial objects, so maybe the problem lies within the use of Serial.write().
Any help would be greatly appreciated. 💃 `#include <Wire.h>
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);

void setup() {
  lcd.begin(16, 2);
  lcd.clear();
  Serial.begin(115200);
}

void loop() {
  Serial.write("?");
  Serial.write("\n");
  lcd.setCursor(0, 0);
  lcd.print(Serial.readString());
  Serial.flush();
  delay(1000);
  lcd.clear();
}`

winder, 2016-06-07:
You need to send `\r\n` after the command you want to send to GRBL.

Krenail, 2016-06-07:
I tried with this code (notice the single ' sign instead of the double " one to test)
`#include <Wire.h>
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);

void setup() {
  lcd.begin(16, 2);
  lcd.clear();
  Serial.begin(115200);
  lcd.setCursor(0, 0);
  lcd.print("ciao");
  delay(1000);
  lcd.clear();
}

void loop() {
  Serial.write("M3");
  Serial.write("\r");
  Serial.write("\n");
  lcd.clear();
  lcd.print("M3 1");
  delay(3000);
  Serial.print("M3");
  Serial.print("\r");
  Serial.print("\n");
  lcd.clear();
  lcd.print("M3 2");
  delay(3000);
  Serial.write("M3");
  Serial.write('\r');
  Serial.write('\n');
  lcd.clear();
  lcd.print("M3 3");
  delay(3000);
  Serial.print("M3");
  Serial.print('\r');
  Serial.print('\n');
  lcd.clear();
  lcd.print("M3 4");
  delay(3000);
}`
but it still doesn't work.

winder, 2016-06-07:
You could test the RX code first by resetting GRBL until you can read startup message with your secondary Arduino. I found this online where someone was doing the same thing:
http://letsmakerobots.com/node/35111

Krenail, 2016-06-07:
I just tried resetting GRBL with a jumper cable between A0 and GND but the only thing that happens is blinking of RX and TX LEDs.

vlachoudis, 2016-06-08:
Remove the 1kOhm resistors, and just use the 3 cables, GND-GND, TX->RX, RX->TX

closed 2016-09-03 by chamnit